cp2004/OctoPrint-EEPROM-Marlin #1: [Meta] Development roadmap/plan for this plugin

cp2004, 2020-09-23:
I will be documenting my initial plans for development of this plugin in this ticket - please don't open issues for features that I already have on this list!

[ ] * Re write plugin to Python server-side
[ ] * UI overhaul
[ ] * Full compatibility to latest Marlin versions

dchauran, 2020-10-09:
"Test compatibility with latest Marlin"

For me at least, this plugin is completely nonfunctional running marlin from bugfix-2.0.x as of last night, and was nonfunctional when loaded from the old repo as well.

When I change a setting and click upload, it runs m500 but I don't see it actually issuing any command to change the settings I changed. When I click backup, the file only contains:

Send: M504

Recv: echo:EEPROM OK
Recv: ok,Send: M501

I can probably fix these issues, but wanted to make sure you aren't actively looking at them before I dive in.

cp2004, 2020-10-12:
Hey @dchauran - the backup problem is tracked in #8, I haven't had a chance to dig deep since I am working through implementing the EEPROM stuff in the virtual printer, so I don't have to develop on my production machine for this one. Its almost there, just needs all of the commands now. I'm not actively looking into them *right now* so if you wanted to take a look, by all means go ahead.

dchauran, 2020-10-12:
thanks, will take the conversation over there

cp2004, 2020-12-13:
Everything on this roadmap is covered for now, as of the release of 3.0.0

Closing, details at #15

closed 2020-12-13 by cp2004