cp2004/OctoPrint-EEPROM-Marlin #2: Remove whitespace before Firmware Info

thelewiss1, 2020-10-04:
Hi there, thanks for taking over the plugin.

I've found some issues to query you about, and i'll keep those seperate.

Please see the image:
[image]

There is a large amount of whitespace in front of the Firmware Info output.

Thanks!

cp2004, 2020-10-04:
I am aware of this one, still trying to dig through the code to understand why it appears! It will be fixed at some point.

thelewiss1, 2020-10-04:
> I am aware of this one, still trying to dig through the code to understand why it appears! It will be fixed at some point.

Great. Thanks!

cp2004, 2020-10-05:
So while I was able to reproduce this initially, when I went to analyse it to fix, I then couldn't get it to happen 🙁 

Going off code analysis, I added in some bits that trim down the whitespace, at any points I think it *could* turn up.

At some point, please try out the development branch, by installing in 'plugin manager > get more > ...from url' then paste `https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/devel.zip` and you should be on the way.

thelewiss1, 2020-10-06:
> So while I was able to reproduce this initially, when I went to analyse it to fix, I then couldn't get it to happen 🙁
>
> Going off code analysis, I added in some bits that trim down the whitespace, at [...]

Hi there!
That's fixed it i believe, i've tried reloading a few times with no preceding whitespace.
Thanks!

cp2004, 2020-10-06:
Excellent news, thank you for testing it!

cp2004, 2020-10-07:
2.0.0 Has been released, hopefully you get the update notification

closed 2020-10-07 by cp2004