cp2004/OctoPrint-EEPROM-Marlin #3: [Feature Request] Z-Offset

KenLucke, 2020-10-07:
When you get into the tasks you have set for yourself, please add the Z-Probe Offset into the parameters that are read/set (this is different, of course, from the simple Z-Offset already included).

cp2004, 2020-10-08:
I can look into this, but may need testing when its implemented before release since I don't have a printer with a probe.

invisiblek, 2020-10-09:
The regex is wrong for newer versions of marlin (2.0.0 and beyond i think, but i need to dig through commit history) so the options just don't show up. They now include probe X/Y offsets too. The whole version detection stuff in this plugin is a shit show though. I've hacked it to work on the old repo (since i was only ever running it on my marlin build), but will need to be retooled to be compatible with all versions. I can look in to this if I get some time.

cp2004, 2020-10-09:
> The whole version detection stuff in this plugin is a shit show though

I can agree with you there (why compare *individual version strings*??!), there are a few things like this that make me scratch my head. 

The last update the plugin saw before I took over maintenance last week, was 2018 (before 2.x was really *a thing*) so it is on my list in #1 to test with the latest versions of Marlin (I have 2.0.5.3 available, and looking at putting EEPROM into the Virtual Printer in OctoPrint so I could simulate this)

Thanks for looking into it!

cp2004, 2020-11-22:
Apologies that this has taken such a long time to release - I am in progress of re-writing this plugin from the ground up, and it has taken longer than I initially thought to get around to, than when I decided not to release a new version before then. Its on its way, the data collection and sending is done, it now needs a UI. Hopefully it will make it before the end of 2020 still. Stay tuned!

cp2004, 2020-12-14:
@KenLucke @invisiblek

Please test the release candidate of V3 of this plugin, it is available and this feature/fix is implemented.

Details can be found in the feedback ticket of how to install, update and report issues: #16

KenLucke, 2020-12-14:
I have tried SEVERAL times to use the URL https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/3.0.0r3.zip to try to install via the Plugin Manager.  All it does is sit there and look at me.  It never installs, never sends notices, refuses to enable or accept the close button... nothing.  I have to reload OP to get out of the install screen.

[image]

I've restarted OP, rebooted the whole system, still unable to install the RC.

cp2004, 2020-12-14:
@KenLucke take a look at the conversation in #16, the URL had a typo but has now been updated. Thanks for giving it a go!

KenLucke, 2020-12-14:
Ok, got it working.  Looks awesome so far, with the exception of the date/time formatting issue mentioned elsewhere.  Much more intuitive.

PR #7 (2020-10-09): probe offsets for marlin 2.0.0+
Fixes #3

* In approximately marlin 2.0.0, M851 became capable of adjusting
  not just Z, but also X and Y probe offsets.
* Create regex to test for that case and if we do indeed have
  X, Y and Z probe offsets to fiddle with, shove it in the "Offsets"
  tab (renamed from "Home offset")

[image]
commit add2c943 (2020-10-09): probe offsets for marlin 2.0.0+

Fixes #3

* In approximately marlin 2.0.0, M851 became capable of adjusting
  not just Z, but also X and Y probe offsets.
* Create regex to test for that case and if we do indeed have
  X, Y and Z probe offsets to fiddle with, shove it in the "Offsets"
  tab (renamed from "Home offset")
closed 2020-12-20 by cp2004