cp2004/OctoPrint-EEPROM-Marlin #4: [Feature Request] - Named backups

KenLucke, 2020-10-07:
I'd love to see named backups that I could load.

The reason is that for highest accuracy, Bed Leveling, z-probe offsets, PID tunes, and a few other things depend on the temperatures to which the nozzle and/or bed are set.  If there were named backups, stored on the local machine (i.e., the browser machine), you could load the one that had your PID tune, z-probe offset, and bed level mesh for the specific temperature you were about to print.

cp2004, 2020-11-24:
Current plan is to implement this as a new section, storing the backups on the server too. Unfortunately backups from the previous version will not be compatible - I might be able to make a way of 'translating' them temporarily. So you could kind of, load different profiles from the UI no need to look after the files properly.

cp2004, 2020-11-24:
Note that this might not make it into the initial release (well, my first proper release) since its quite complex. Basic backup will be there though. Depends on how good I am at designing the UI....

cp2004, 2020-12-13:
Implemented with the above commit, as part of an all-new backup feature.

Ready for 3.0.0

KenLucke, 2020-12-14:
Working with RC3 right now, but not seeing how to set, edit, or change names.  Did it make it into this release?

cp2004, 2020-12-14:
Head to the settings page, then check 'use custom names'. Once that is checked, you will be prompted to name the backup as it is created.

KenLucke, 2020-12-14:
Ah, got it.  I had gotten used to there not being anything on the settings page :)

KenLucke, 2020-12-14:
Beautious.

commit 4e9690c0 (2020-12-13): :sparkles: Support naming backups as they are created

closes #4
closed 2020-12-20 by cp2004