cp2004/OctoPrint-EEPROM-Marlin #5: Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

jreed509, 2020-10-08:
**Describe the bug**
Error while installing through plugin manager in Octoprint. 

When installing the plugin, I receive the following pip error: "Could not parse output from pip....". Installation using the old link from `ambsr` works properly.

**Logs**
Installing plugin from https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpKzaJWr/OctoPrint-EEPROM-Marlin-master.zip --no-cache-dir
Processing /tmp/tmpKzaJWr/OctoPrint-EEPROM-Marlin-master.zip
Requirement already satisfied (use --upgrade to upgrade): Marlin-EEPROM-editor==2.0.0 from file:///tmp/tmpKzaJWr/OctoPrint-EEPROM-Marlin-master.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: semantic_version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: futures<4,>=3.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: monotonic<2,>=1.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: scandir<2,>=1.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: chainmap<2,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: typing<4,>=3.7.4.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: singledispatch in /home/pi/oprint/lib/python2.7/site-packages (from tornado==5.1.1->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: backports_abc>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from tornado==5.1.1->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python2.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python2.7/site-packages (from rsa==4.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<1,>=0.10.2->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
[... cut ...]
Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->Marlin-EEPROM-editor==2.0.0)
**Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output**

**OctoPrint:**
Operating System running OctoPrint: Raspberry Pi 3B
Octoprint Version:  1.4.2
**Printer model & firmware version:**
Browser: Chrome

**Additional context, screenshots, anything else you think might be useful**
Add any other context about the problem here.

cp2004, 2020-10-08:
Have you got the `plugin_pluginmanager_console` log file? From that bit, there is no error so hopefully that should show it. Don't think it's a general issue since OctoPrint's tracking shows quite a few installs of my version.

jreed509, 2020-10-08:
Sure, here's the `plugin_pluginmanager_console.log` output. Unfortunetly, it looks pretty much the same as above:

```
2020-10-06 19:26:19,852   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmp_YWaxF/master.zip --no-cache-dir
2020-10-06 19:26:22,883 > Processing /tmp/tmp_YWaxF/master.zip
2020-10-06 19:26:27,962 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,963 > Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,963 > Requirement already satisfied: uritools<2.2,>=2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,964 > Requirement already satisfied: SQLAlchemy<1.2,>=1.1.15 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,964 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,964 > Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,965 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,965 > Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,965 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,966 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,966 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,967 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,967 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,968 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,968 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,969 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,969 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,969 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,970 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,971 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,971 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,971 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,972 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,972 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,973 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,973 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,973 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,974 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
2020-10-06 19:26:27,974 > Requirement already satisfied: semantic_version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-FilamentManager==1.6.0)
[... cut ...]
2020-10-08 13:37:54,860 > Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
2020-10-08 13:37:54,861 > Requirement already satisfied: chardet<4,>=3.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Marlin-EEPROM-editor==2.0.0)
2020-10-08 13:37:54,861 > Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->Marlin-EEPROM-editor==2.0.0)
```

cp2004, 2020-10-08:
So it looks to me like everything was successful. Picking it apart:
* You installed the new version from me (`2.0.0`), successfully:
   `2020-10-07 17:17:45,903 > Successfully installed Marlin-EEPROM-editor-2.0.0`
* Then you tried to uninstall the old one (since it seems that didn't happen automatically) I have figured this one out, and it is my fault. I changed the name in `setup.py`  to bring it in line with the actual name, and it seems that doing that means that pip doesn't find the old version to uninstall.
* After that, you kept trying to install the new one, but it was already installed so it just said `requirement already satisfied`

cp2004, 2020-10-08:
Now I've just got to figure out how is best to fix this... 🤦 since it will break automatically updating for some people

cp2004, 2020-10-08:
If you haven't got it working, please try and uninstall whatever EEPROM plugin is there, then install it again from the plugin manager. I have updated the wording in the notice to hopefully avoid this

jreed509, 2020-10-09:
That worked for me. I needed to manually uninstall v2.0.0 through pip (over ssh) and then reinstall the latest version from the plugin manager. 

Everything seems to be working properly now. Thanks!

closed 2020-10-09 by jreed509