cp2004/OctoPrint-EEPROM-Marlin #6: [Request]Touchui interface icon for plugin

devildant, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
with plugin touchui the tab EEPROM editor use "?" icon
[image]

**Describe the solution you'd like**
use this css inside your plugin 
#touch body #tabs #tab_plugin_eeprom_marlin_link a:before, #touch body #navbar #all_touchui_settings #tab_plugin_eeprom_marlin_link a:before, #touch body #tabs #tab_plugin_eeprom_marlin_link a:before, #touch body #navbar #all_touchui_settings #tab_plugin_eeprom_marlin_link a:before {
    content: "\f14b";
}

it used this icon : https://fontawesome.com/v4.7.0/icon/pencil-square

**Describe alternatives you've considered**

**Additional context**

cp2004, 2020-10-08:
Seems doable, on the list. Unless you want to make a PR, since you already know how to fix it....? 😉

devildant, 2020-10-08:
Both :p 
I communicate the CSS trick to all the plugins I use: p
and I'm in the middle of dev currently I don't have much time to do a pull request, but I am sending the solution :)

devildant, 2020-10-08:
I made a small typo :p my basic message was edit (test in the dirty by injecting directly into the html, it's ok) :p

cp2004, 2020-10-17:
Ready for 2.1.0

cp2004, 2020-11-22:
Apologies that this has taken such a long time to release - I am in progress of re-writing this plugin from the ground up, and it has taken longer than I initially thought to get around to, than when I decided not to release a new version before then. Its on its way, the data collection and sending is done, it now needs a UI. Hopefully it will make it before the end of 2020 still. Stay tuned!

cp2004, 2020-12-14:
@devildant 

Please test the release candidate of V3 of this plugin, it is available and this feature/fix is implemented.

Details can be found in the feedback ticket of how to install, update and report issues: #16

commit 36b0a631 (2020-10-17): :lipstick: Add TouchUI icon

Closes #6
closed 2020-12-20 by cp2004