cp2004/OctoPrint-EEPROM-Marlin #8: [Bug] Backup Feature Doesn't Backup

DodgeDeBoulet, 2020-10-10:
**Describe the bug**
Perhaps I' m misunderstanding its function, but the Backup button seems to create a file that contains nothing useful. It certainly doesn't contain human-readable GCode for restoring EEPROM settings, which is what I would expect it to do.

**Steps To Reproduce**
1. Load your firmware's settings
2. Click on Backup
3. Save the downloaded file when prompted
4. Examine file's contents

Mine:
```
Send: M504
Recv: echo:EEPROM OK
Recv: ok P63 B31,Send: M501
```

**Expected behavior**
Not sure what to expect, but I wasn't expecting this.

**Logs**
[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5359678/octoprint.log)

**Versions, system information**
**Plugin**:
2.0.0
**OctoPrint**
1.4.2
**Operating System running OctoPrint**:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

Linux octopi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

Printer model & firmware version:
Ender 5 Plus, BTT SKR 1.4 Turbo, Marlin 2.0.6.1

Browser:
Microsoft Edge Chromium

**Additional context, screenshots, anything else you think might be useful**

cp2004, 2020-10-10:
Bearing in mind I didn't write this, to my understanding it is just saving a log of what it receives from an M501 command. Looks like that might have been lost in yours.

Mine looks a bit like this (when I tried it out):
```gcode
Send: M504
Recv: echo:EEPROM OK
Recv: ok,Send: M501
Recv: echo:V76 stored settings retrieved (640 bytes; crc 43039)
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:; Steps per unit:
Recv: echo: M92 X100.00 Y100.00 Z400.00 E90.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X400.00 Y400.00 Z4.00 E50.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z1000.00 E1000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
```

and it does look like it has cut off the acceleration from my backup too. 

Maybe you could try and see what the full output of M501 is?

I will try and look into it at some point, but I don't particularly like the way it is backing up anyway....

DodgeDeBoulet, 2020-10-10:
Full output from M501:
```
Send: M501
Recv: ==onLoadSettings==
Recv: echo:V81 stored settings retrieved (710 bytes; crc 47994)
Recv: Unified Bed Leveling System v1.01 active
Recv: 
Recv: Unified Bed Leveling initialized.
Recv: 
Recv: Mesh loaded from slot 0
Recv: Mesh 0 loaded from storage.
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E136.84
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X750.00 Y750.00 Z10.00 E75.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000.00 Y2000.00 Z100.00 E75.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P750.00 R1000.00 T300.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: 
Recv: echo:; Active Mesh Slot: 0
Recv: echo:; EEPROM can hold 12 meshes.
Recv: 
Recv: echo:; PID settings:
Recv: echo:  M301 P27.08 I2.51 D73.09
Recv: echo:  M304 P131.06 I11.79 D971.23
Recv: echo:; Retract: S<length> F<units/m> Z<lift>
Recv: echo:  M207 S3.00 W13.00 F2700.00 Z0.00
Recv: echo:; Recover: S<length> F<units/m>
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:; Auto-Retract: S=0 to disable, 1 to interpret E-only moves as retract/recover
Recv: echo:  M209 S0
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-45.00 Y-5.00 Z-4.44
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X550 Y550 Z550
Recv: echo:  M906 I1 Z550
Recv: echo:  M906 T0 E660
Recv: 
Recv: echo:; Hybrid Threshold:
Recv: echo:  M913 X152 Y152 Z10
Recv: echo:  M913 I1 Z10
Recv: echo:  M913 T0 E48
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 I1 Z
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.01
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L100.00 U75.00
Recv: echo:; Backlash compensation:
Recv: echo:  M425 F0.00 X0.00 Y0.00 Z0.00
Recv: ==onConfigurationStoreRead==
Recv: 
Recv:  init zprobe_zoffset = -4.44
Recv: ok P63 B31
```

cp2004, 2020-10-10:
Something's broken with it capturing the settings. 🤔 will take a look, and see what I can do. Thank you for the detailed report!

dreiduke, 2020-10-11:
same here 
this is the backup file....  not more not less

Send: M504

Recv: echo:EEPROM OK
Recv: ok,Send: M501

dchauran, 2020-10-12:
@cp2004 I am going to take a look at fixing this one. A few random thoughts/questions:

I don't think we should be using M501 by default. M501 reloads the saved eeprom settings, potentially overwriting any tweaks to the running settings. M503 prints active (not necessarily stored) settings, and also prints out a usable gcode script, whereas M501 adds extra stuff at the top which needs to be parsed out. I would propose that we use M503, and give the user an option e.g.

What to backup: 
* Running settings.
* Stored eeprom settings. Warning, any changes made since eeprom was last saved (e.g. via M500 or "upload") will be lost.

If they select running settings, display a Checkbox (checked by default) that says "overwrite stored eeprom settings" which triggers M500, then M503. If they select stored eeprom settings, we run M501 then M503.

As to the method of taking the backup, I do not see much of an option other than generating a gcode script. Marlin to my knowledge doesn't provide any other means of setting eeprom configuration (e.g. no way to upload a settings file).

cp2004, 2020-10-12:
Sounds right to me 🙂

I was thinking of making it an option in the settings page (I'll get rid of the clutter out of there) to use M501 *or* M503 to load settings, since Marlin can have M503 disabled in config.h. Thank you in advance for the contribution!

cp2004, 2020-10-17:
@dchauran - Just a heads up if you are still yet to look into this, you might want to pull the latest changes on the `devel` branch. I've run OctoPrint's pre-commit suite against this plugin, including black and prettier formatting, so merges will be easier if you pull first. If you've already started a fix I'm sure we can make a merge work just fine.

See the changes here: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/commit/7f84dfa90580427290c0fafaec681e330f33b84c

dchauran, 2020-10-20:
Thanks, I was out of town, so had only done a little bit of poking. Will pull devel into my fork.

The original backup/restore code seems to be more complex than necessary, so it's taking some thought in how to best handle this.

See the changes here: 7f84dfa<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fcp2004%2FOctoPrint-EEPROM-Marlin%2Fcommit%2F7f84dfa90580427290c0fafaec681e330f33b84c&data=04%7C01%7C%7Cfd2389f9e8554d97176808d872bcd490%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637385502629241223%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=ZGHdIC1WxEUW%2FgKD2xk8qlXjWfa8wFbwzW2n9pFK4Lg%3D&reserved=0>

cp2004, 2020-10-20:
No worries, I haven't had a chance to look either!
Trying to balance working evenly across this other OctoPrint projects, all in my spare time. I can wait as long as it takes.... 🙂

cp2004, 2020-11-22:
@dchauran If you are still looking in to this, don't worry about it. I am in progress of re-writing this plugin from the ground up, and it will be really simple to create backups the way I have implemented the data structures, so it will work very well. Targeting before end of 2020 for release, this will be a *much* improved plugin.

cp2004, 2020-12-13:
The backup feature is ready and implemented for 3.0.0. Summary of features:
* Create a backup on the server as a JSON file from the internal data of the plugin
* Restore these backups at any time, right from the UI
* Upload a backup
* Download a backup
* Custom backup names
* Timestamps in the backups

cp2004, 2020-12-14:
@DodgeDeBoulet @dreiduke @dchauran 

Please test the release candidate of V3 of this plugin, it is available and this feature/fix is implemented.

Details can be found in the feedback ticket of how to install, update and report issues: #16

dchauran, 2020-12-15:
Hi there - Unfortunately, my printer decided to stop working correctly some months ago and I have not gotten around to fixing it, so I am unable to test at this time.

cp2004, 2020-12-20:
3.0.0 has been released 🎉

closed 2020-12-20 by cp2004