cp2004/OctoPrint-EEPROM-Marlin #9: [Bug] Cosmetic error only: Shows vague error message while printing

KenLucke, 2020-10-19:
**Describe the bug**
Cosmetic bug - error message is somewhat confusing.

**Steps To Reproduce**
Start a print
Open Editor tab shows this error, despite the editor working just fine before starting the print:
[image]
This could confuse someone into thinking that the plugin won't work for them or that there's been some sort of massive sudden EEPROM fault, when, in fact, it works fine if they were not printing at the moment.

Tiny little thing in the larger order of things, but I thought it was worth pointing out.

**Expected behavior**
Should say something on the order of "Error: Either this plugin does not function with your machine (only Marlin based EEPROM machines are supported) , or the plugin is currently disabled because a print is in progress."

**Logs**

`octoprint.log`:  not needed for this one.

**Versions, system information**
**Plugin**:
Version:  2.0
**OctoPrint**
Version: 1.4.2
Operating System running OctoPrint:
Mac OS X 10.15.7
Printer model & firmware version: n/a
Browser: n/a

**Additional context, screenshots, anything else you think might be useful**

cp2004, 2020-10-24:
Looks like that's actually the two different error messages *at the same time*, they should be two different cases. Think this happens if you start a print, then load the UI later on. Will fix.

cp2004, 2020-11-30:
Fixed for the next release. Error messages include:
Disconnected:
[image]
Read-only mode:
[image]
Firmware not marlin:
[image]

And maybe more, I can't remember!

cp2004, 2020-12-20:
3.0.0 has been released 🎉

closed 2020-12-20 by cp2004