cp2004/OctoPrint-EEPROM-Marlin #10: [Bug] Not working after update to Marlin 2.0.7.2

jjvelar, 2020-11-01:
Hi,
I have an Anet A8. With old firmware SkyNet3D, 2.0.0 version of this pluging was working perfectly well but after updating to Marlin 2.0.7.2, all tabs have disappeared and only can see some firmware values.
Here you've got a screen capture to illustrate.
I had to disable a lot of stuff in the firmware to keep auto level and power loss functionality, so I have also attached my configuration.h and configuration_adv.h files in case this is what is causing the issue.
Octoprint log is also attached.

[Configuration.zip](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5471364/Configuration.zip)
[image]

Thanks and best regards,

José
[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5471390/octoprint.log)

cp2004, 2020-11-02:
Had a look at the configuration files, they seem OK (the relevant parts are still enabled).

Could you try a couple of things, so I can try and see the source of the issue:
1. Please try pressing 'load' to read the settings from the printer. This may not have happened automatically and is something I am working on.
2. If this does not work, please enable 'Serial Logging' (Settings > Serial connection), then press 'load', then upload the `serial.log` file here afterwards. This will allow me to see what is communicated to the printer.

jjvelar, 2020-11-02:
Hi,
Thanks for your fast reply.
I have tried to press load and it reads all settings but the other tabs that allow to modify values do not appear.
Here you've got the serial log file.
Best regards,

José
[serial.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5477874/serial.log)

cp2004, 2020-11-02:
Ahh I see the issue. Seems the plugin needs 'EEPROM Chitchat' to be enabled (`Configuration.h, L1483`). The plugin uses M501 to get the data, rather than M503 - to be changed in a future update.

If you don't have enough storage for that option, then until I can get a chance to work on the next release, there is no other option.

jjvelar, 2020-11-02:
Oh, I see. That makes sense.
I don’t have enough memory so... do you think you can fix this in the short
term or not really?
Thanks!

cp2004, 2020-11-02:
I'm re-writing *most* of this plugin, working on and off when I can. Some parts of it are very hard to extend, so they are being remade. It's difficult to know when it will get done. Not soon really... Hopefully in a month or so it will be nearly ready to work.

You could try swapping M503 support (disable that to release memory) for EEPROM chitchat? But then it depends which is more useful.

jjvelar, 2020-11-02:
I have already disabled M503.
Not a lot of available space in my Anet A8 original board so I guess I will
have to wait.
Thanks anyway

José

jjvelar, 2020-11-03:
Hi,
I tried after activating EEPROM chitchat but still the same issue.
Here you've got what I've got in configuration.h:

#define EEPROM_SETTINGS       
#define DISABLE_M503        
#define EEPROM_CHITCHAT       
#define EEPROM_BOOT_SILENT    
#if ENABLED(EEPROM_SETTINGS)
  //#define EEPROM_AUTO_INIT  
#endif

Any other suggestion?
Thanks and best regards,

José

jjvelar, 2020-11-21:
Hi @cp2004 
Any news on this?
I have activated EEPROM_CHITCHAT but it still does not work so don't know if the bug is due to this...
Thanks and best regards,

José

cp2004, 2020-11-22:
I was quite sure that that was the setting that you needed to change. Still no response on an `M501` command? I asked in the Marlin Discord, but no one responded. Will play about with this and test it myself later today I guess then.

Re-write progress is good, but it will still require the settings to be outputted on *either* and `M501` or `M503` command.

jjvelar, 2020-11-22:
Hi,
No, no response on M501 command.
Let me know if you want me to test anything.
Best regards,

José

cp2004, 2020-11-23:
@jjvelar So according to a helpful person (namely @sjasonsmith, sorry if that ping was annoying) you need to have `DISABLE_M503` commented as well. This is in the Marlin code:
```c++
#if ENABLED(EEPROM_CHITCHAT) && DISABLED(DISABLE_M503)
      // Report the EEPROM settings
      if (!validating && TERN1(EEPROM_BOOT_SILENT, IsRunning())) report();
#endif
```
which shows you need them both to be enabled for M501 command. Hope that helps

jjvelar, 2020-11-24:
oh, I see. I will try it later.
thanks for your support.
best regards,

José

jjvelar, 2020-11-25:
Hi,
It is now working but "Advanced" section is not available (therefore I can't change, as an example, Z offset)
It was available when I was using SkyNet3D firmware instead of Marlin 2.x.
Is there any other option I have to activate to get access to this section?
Thanks and best regards,

José

[image]

cp2004, 2020-11-25:
The items don't show up unless it can detect some data out from them. So you will need to look in the terminal when you send `M501` command, and look for the data. Maybe there is something slightly wrong with the regex in the plugin, and it can't parse the response fully.

I can't put any work into debugging this at the moment, since the parsing layer is brand new in the next release, and works *far far* better.

jjvelar, 2020-11-26:
Hi,
If you need a beta tester for the new plugin, just let me know.
This is what I get after sending M501:

IGIPOTS LoadingDIGIPOTS LoadedDIGIPOTS LoadingDIGIPOTS Loadedecho:V82
stored settings retrieved (669 bytes; crc 13871)echo:  G21    ; Units
in mm (mm)echo:  M149 C ; Units in Celsiusecho:; Steps per unit:echo:
M92 X101.60 Y101.60 Z401.20 E100.00echo:; Maximum feedrates
(units/s):echo:  M203 X400.00 Y400.00 Z8.00 E50.00echo:; Maximum
Acceleration (units/s2):echo:  M201 X2000.00 Y2000.00 Z100.00
E10000.00echo:; Acceleration (units/s2): P<print_accel>
R<retract_accel> T<travel_accel>echo:  M204 P4000.00 R1000.00
T4000.00echo:; Advanced: B<min_segment_time_us> S<min_feedrate>
T<min_travel_feedrate> J<junc_dev>echo:  M205 B20000.00 S0.00 T0.00
J0.10echo:; Auto Bed Leveling:echo:  M420 S0 Z0.00echo:  G29 W I0 J0
Z0.22632echo:  G29 W I1 J0 Z0.14158echo:  G29 W I2 J0 Z0.17647echo:
G29 W I0 J1 Z0.03689echo:  G29 W I1 J1 Z0.01944echo:  G29 W I2 J1
Z-0.01296echo:  G29 W I0 J2 Z0.40828echo:  G29 W I1 J2 Z0.32104echo:
G29 W I2 J2 Z0.15653echo:; Material heatup parameters:echo:  M145 S0
H200 B0 F0echo:  M145 S1 H240 B90 F0echo:; PID settings:echo:  M301
P29.34 I1.98 D108.74echo:  M304 P295.00 I35.65 D610.21echo:; Z-Probe
Offset (mm):echo:  M851 X-26.00 Y-40.00 Z-0.90echo:; Filament
load/unload lengths:echo:  M603 L0.00 U100.00

cp2004, 2020-12-14:
@jjvelar 

Please test the release candidate of V3 of this plugin, it is available for you to test now!

Details can be found in the feedback ticket of how to install, update and report issues: #16

jjvelar, 2020-12-15:
Hi

New version v3 RC3 works for me but I still miss Z Probe Offsed value and
the possibility to change it.
This field was available in "Advanced" version when I was using Skynet3D
instead of Marlin 2.0.7.
Any idea about this? By the way, the new version looks good. Thanks for
doing this.
Best regards,

José

jjvelar, 2020-12-15:
Forget my last email as I have found it. it is under "Offset". Of course...
Thanks!!!!!

cp2004, 2020-12-20:
3.0.0 has been released 🎉

closed 2020-12-20 by cp2004