cp2004/OctoPrint-EEPROM-Marlin #11: [Bug] Minor bug, notification on install

cr0wm4n, 2020-11-07:
when i installed this and i needed to restart, the notification told me ws2812b was successfully installed

cp2004, 2020-11-07:
I'm slightly confused about this, since installing this plugin shouldn't show that message, and I can't find any copy-paste errors in this plugin, since the WS281x one is also mine. Can you upload the `plugin_pluginmanager_console.log` (I think that should be what it is called) from when you installed this plugin

cr0wm4n, 2020-11-07:
I cant find that log :(
so i removed it and reinstalled it to try and get a screenshot, but this reinstall the notification is correct. Wierd!!!

cr0wm4n, 2020-11-07:
its back...
After i clicked reload on the new install and your announcement comes up its when i click the finish button..
[image]

cp2004, 2020-11-07:
Hmm, that's from the latest update in WS281x LED Status, supposed to be when you press 'finish'. I will have to look at that, maybe it is an OctoPrint issue - I think I know what is happening. Since that should not be triggered unless the *ws281x led status* VM is bound. Transferring this issue to that repository since it is more related.