cp2004/OctoPrint-EEPROM-Marlin #12: [Bug]

TomW1605, 2020-11-16:
**Describe the bug**
intermittent issue with marlin printer.

**Steps To Reproduce**
1. Go to the EEPROM tab
2. See error. this is intermittent and i have not pined down the cause yet.

**Expected behavior**
it works

**Logs**

`octoprint.log`: https://pastebin.com/eVZZWukX (i have remove the server heartbeat entries because the log was too big for pastebin)

**Versions, system information**
**Plugin**:
Version: 2.0.0
**OctoPrint**
Version: 1.5.0rc1
Operating System running OctoPrint: OctoPi

Printer model & firmware version: Prusa i3, Marlin
Browser: Chrome

**Additional context, screenshots, anything else you think might be useful**

cp2004, 2020-11-16:
I think it happens if you reload the page mid print, since it cannot send the necessary commands to get the info. Vague error messages are also a factor in #9.

I know I keep saying this, but I am midway through a re-write of this plugin, that eliminates these issues by running the data-collection and processing server side. There were a few things ahead of it on the list but I am hoping to pin down some time at the end of this week to really dig into it.

TomW1605, 2020-11-17:
this happens even when i have not printed all day so i dont think it is that.

cp2004, 2020-11-18:
When this happens, try sending an `M115` command to the printer. Think the problem is that the plugin does not do anything when you reload the page, *after* the printer is connected. If you send that command and it wakes up, you should then be able to press the buttons to do the right things.

TomW1605, 2020-11-19:
ok i tested that and yes as soon as i send `M115` it works. might be worth adding a refresh button and slightly altering the error message.

cp2004, 2020-11-19:
> might be worth adding a refresh button and slightly altering the error message

Hehe, I'm going beyond that, writing this issue out entirely. I agree that would be the way to go if I was not re-writing the plugin, but that's exactly what I'm doing 🙂

TomW1605, 2020-11-20:
that works too thanks for letting me know.

closed 2020-11-20 by TomW1605