cp2004/OctoPrint-EEPROM-Marlin #13: [Bug] Stuck on Hello dialog

reggaemanu, 2020-11-25:
**Describe the bug**

After installing I'm stuck at the "hello from a new maintainer" dialog. Impossible to close it, the exit button does nothing, so I can't use octoprint anymore.

**Steps To Reproduce**
1. Uninstall the old plugin
2. Install the new version
3. Get screwed

**Expected behavior**
Honestly I expected that exit button to just works.

**Logs**
I don't have access to octoprint anymore as I'm stucked on this dialog

**Versions, system information**
**Plugin**:
Version: latest version
**OctoPrint**
Version: 1.5.0rc3
Operating System running OctoPrint: octopi

Printer model & firmware version: doesn't matter
Browser: Google chrome
[image]

cp2004, 2020-11-25:
> I don't have access to octoprint anymore as I'm stucked on this dialog

New solution in OctoPrint 1.5.0 (since you seem to be running the RCs) - the recovery page.

Head to `http://<yourip>/recovery` then you should be able to start OctoPrint in safe mode, delete the plugin and then restart OctoPrint and become 'un-screwed'

In the meantime, I will investigate why this happens. There is nothing in this plugin that can stop the wizard closing (literally only HTML, no script in that one) so I need to investigate this upstream.

To do this, I have a feeling it might be configuration-specific. Could you upload a **redacted** config.yaml (ie. take out API keys, unique tokens etc. - anything that looks like it might be risky sharing. There shouldn't be too much, but definitely API key.). This would really help, since I don't think

cp2004, 2020-11-27:
@reggaemanu Any news on whether you've been able to look at this? I would like to know if it is a potential bug in 1.5.0rc series, which is planned for next week. I have only seen this twice, and both times they were on 1.5.0rc, so it would be great if we could look into this.
Thanks!

reggaemanu, 2020-11-27:
Hi,

Yeah I managed to unstuck this with the recovery mode today (really handy
this new feature), but I didn't have time to retrieve the yalm config file
for you before leaving for my workshop.
I'll send you that this evening when coming back home.

cp2004, 2020-11-27:
No worries, thanks for getting back to me!

reggaemanu, 2020-11-27:
Here's my config.yaml file
[config.txt](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5609430/config.txt)

cp2004, 2020-11-27:
Thank, I will take a look and see if we can find something.

cohaolain, 2020-11-28:
For what it's worth, though this may well have been found already! The following messages are logged to console on each attempt to close the wizard dialog.
[image]

cp2004, 2020-11-29:
@cohaolain could I get an OctoPrint.log please? Internal server error is definitely something that needs investigation.

cohaolain, 2020-11-29:
Hey @cp2004 I'll try take a look at this in a bit :slightly_smiling_face:

cohaolain, 2020-11-29:
I just tried to reproduce in a clean config, containerised instance - couldn't reproduce it this time. So likely either
1. a compatibility issue with the update (in the sense that having a config generated by an older version is causing issues) 
and/or
2. a plugin conflict - I have quite a few plugins in my production octoprint instance.

I'll see if my production octoprint log has anything useful in it, unsure what my verbosity was set to last time (and can't try to reproduce right now, currently mid-print :smile: )

cohaolain, 2020-11-29:
Here's all the relevant log entries related to the above:
```
2020-11-28 22:41:10,222 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 387, in setSettings
    response = _saveSettings(data)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 933, in _saveSettings
    "gcode", name, script.replace("\r\n", "\n").replace("\r", "\n")
AttributeError: 'NoneType' object has no attribute 'replace'
2020-11-28 22:41:10,224 - tornado.access - ERROR - 500 POST /api/settings (172.23.0.1) 5.32ms
```

cp2004, 2020-11-29:
Thanks for the insight, I will try and take a look. This doesn't look fun, and must be configuration specific.... 😢

cp2004, 2020-11-30:
Fix for this is ready for OctoPrint 1.6.0, (**not** 1.5.0) since it was not a regression. 
Details: https://github.com/OctoPrint/OctoPrint/commit/ba27cdfd9aff8884daca52a61ced30b56ca9a644

Closing this issue since the root cause has been fixed & I'm not to blame, and in the next version of the plugin the wizard dialog will be gone anyway.

closed 2020-11-30 by cp2004