cp2004/OctoPrint-EEPROM-Marlin #14: [Bug] "Hello from a new maintainer!" dialog does not go away

DreadPirateMiko, 2020-11-30:
**Describe the bug**
I've installed the new version on my OctoPi
when loading the Web UI I get the "Hello from a new maintainer!" dialog - clicking "Finish" does nothing.
The log contain the following error:

2020-11-30 20:33:31,481 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 388, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 936, in _saveSettings
    "gcode", name, script.replace("\r\n", "\n").replace("\r", "\n")
AttributeError: 'NoneType' object has no attribute 'replace'

cp2004, 2020-11-30:
Duplicate of #13 
@DreadPirateMiko Please post the full log, since we have so far been unable to reproduce.
/cc @foosel - there are not many plugins that use wizard dialogs, and I can't reproduce it... Must be seriously configuration specific.

DreadPirateMiko, 2020-11-30:
Attaching (from lastets restart) with IP and username redacted 

one thing to mention - I have multiple instances running hence the base url is "http://192.168.1.5xx/**FTi3**/"
```
2020-11-30 23:21:43,921 - octoprint.startup - INFO - ******************************************************************************
2020-11-30 23:21:43,924 - octoprint.startup - INFO - Starting OctoPrint 1.5.0
2020-11-30 23:21:43,926 - octoprint.startup - INFO - ******************************************************************************
2020-11-30 23:21:43,992 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2020-11-30 23:21:43,993 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2020-11-30 23:21:43,994 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2020-11-30 23:21:44,536 - octoprint.startup - INFO - Blacklist processing done, adding 10 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8)
2020-11-30 23:21:44,580 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint2/plugins and installed plugin packages...
2020-11-30 23:21:46,402 - octoprint.plugin.core - INFO - Plugin STL Viewer (0.4.1) is disabled.
2020-11-30 23:21:46,646 - octoprint.plugin.core - INFO - Plugin Yamlpatcher (0.1.1) is disabled.
2020-11-30 23:21:46,865 - octoprint.plugin.core - INFO - Plugin System Command Editor (0.3.2) is disabled.
2020-11-30 23:21:46,923 - octoprint.plugin.core - INFO - Plugin Slicer (1.4.2) is disabled.
2020-11-30 23:21:46,959 - octoprint.plugin.core - INFO - Plugin Slic3r (1.1.1) is disabled.
2020-11-30 23:21:47,335 - octoprint.plugin.core - INFO - Plugin M117NavBar (0.1.0) is disabled.
2020-11-30 23:21:47,352 - octoprint.plugin.core - INFO - Plugin Gcodebar Plugin (0.1.4) is disabled.
2020-11-30 23:21:47,888 - octoprint.plugin.core - INFO - Plugin Dashboard (1.8.0) is disabled.
2020-11-30 23:21:47,921 - octoprint.plugin.core - INFO - Plugin BLTouch Plugin (0.2.0) is disabled.
2020-11-30 23:21:47,939 - octoprint.plugin.core - INFO - Plugin Autoselect Plugin (0.2.0) is disabled.
2020-11-30 23:21:47,980 - octoprint.plugin.core - INFO - Plugin Automatic Shutdown (0.1.4) is disabled.
2020-11-30 23:21:48,428 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2020-11-30 23:21:48,624 - octoprint.startup - INFO - Added settings overlay from plugin consolidate_temp_control
2020-11-30 23:21:48,636 - octoprint.plugin.core - INFO - Found 57 plugin(s) providing 46 mixin implementations, 85 hook handlers
2020-11-30 23:21:49,512 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-11-30 23:21:49,548 - octoprint.server - INFO - Intermediary server started
2020-11-30 23:21:49,595 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint2/uploads...
2020-11-30 23:21:50,622 - octoprint.filemanager - ERROR - Exception while retrieving additional extension tree entries from hook marlinbft
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 95, in full_extension_tree
    hook_result = hook()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlinbft/__init__.py", line 181, in on_get_extension_tree
    accept = self._settings.get(Setting.AcceptExtensions)
AttributeError: 'MarlinbftPlugin' object has no attribute '_settings'
2020-11-30 23:21:50,688 - octoprint.filemanager - ERROR - Exception while retrieving additional extension tree entries from hook marlinbft
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 95, in full_extension_tree
    hook_result = hook()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlinbft/__init__.py", line 181, in on_get_extension_tree
    accept = self._settings.get(Setting.AcceptExtensions)
AttributeError: 'MarlinbftPlugin' object has no attribute '_settings'
2020-11-30 23:21:50,725 - octoprint.filemanager - ERROR - Exception while retrieving additional extension tree entries from hook marlinbft
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 95, in full_extension_tree
    hook_result = hook()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlinbft/__init__.py", line 181, in on_get_extension_tree
    accept = self._settings.get(Setting.AcceptExtensions)
AttributeError: 'MarlinbftPlugin' object has no attribute '_settings'
2020-11-30 23:21:50,763 - octoprint.filemanager - ERROR - Exception while retrieving additional extension tree entries from hook marlinbft
Traceback (most recent call last):
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlinbft/__init__.py", line 181, in on_get_extension_tree
    accept = self._settings.get(Setting.AcceptExtensions)
AttributeError: 'MarlinbftPlugin' object has no attribute '_settings'
2020-11-30 23:21:54,972 - octoprint.filemanager - ERROR - Exception while retrieving additional extension t
```

foosel, 2020-11-30:
That looks like an entirely different problem, for some reason `self._settings` is not being injected? 🤔

cp2004, 2020-11-30:
And that one definitely *isn't* my problem... You know what I might just remove the wizard dialog and push out a patch release, annoying since V3 of this plugin is *so damn close* to being ready for release.

foosel, 2020-11-30:
That's not behind any 500 errors on the settings API or the Wizard though. That's an unrelated exception properly caught as far as I can see (though it is a bug in the plugin)

BSmith-66, 2020-12-01:
i have the same problem

foosel, 2020-12-01:
@BSmith-66 I'm pretty sure @cp2004 would like to see an `octoprint.log` from you as well (as would I).

BSmith-66, 2020-12-01:
sorry without logging in not sure how to get the logs? 
saw other issues that its fixed upstream but not sure how to get in to do any updates etc. 
I can reboot from the recovery screen that's about it 
i am running a Pi0w so its not really supported maybe its time to change that.

foosel, 2020-12-01:
You should also be able to restart in safe mode from the recovery screen.

> i am running a Pi0w so its not really supported maybe its time to change that.

In any case: Yes.

BSmith-66, 2020-12-01:
Figured out I could get the logs by starting safe mode 
i think i have attached them correctly 

[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/5622733/octoprint.log)

cp2004, 2020-12-01:
Not seeing the same issue in that log. It feels to me like it *might* be a regression, but then I haven't had this reported on my other plugins that use wizard dialogs. Maybe I will try setting up the same environment and test. 

Until then, I apologise for the issues, the wizard is gone in future versions, everything in the current development branch works (except the reset button & backup feature) and it is *so close* to release I don't think I have the time/effort to push out a patch release to remove it now. Maybe instead I will prepare it into a 'beta' version people could install.

BSmith-66, 2020-12-02:
Fair call on waiting. 
I just disabled plugin using safe mode for now Im not a heavy user life gets in the way of that.
I just wanted to add a sample to the mix as it may help you if more than 1 user has an issue.
Thanks for the effort.

cp2004, 2020-12-14:
@DreadPirateMiko @BSmith-66 @cohaolain @reggaemanu

Please test the release candidate of V3 of this plugin, it is available and this feature/fix is implemented.

Details can be found in the feedback ticket of how to install, update and report issues: #16

DreadPirateMiko, 2020-12-14:
Can confirm it is now working 
thanks!

Mixdiver, 2020-12-16:
I have the same issue where I can't close the wizard when clicking Finish button. I get this error in Chrome console:

packed_libs.js?ab739628:1 POST http://10.0.0.188/api/settings 500 (INTERNAL SERVER ERROR)
send @ packed_libs.js?ab739628:1
ajax @ packed_libs.js?ab739628:1
OctoPrintClient.ajax @ packed_client.js?8f609640:7
OctoPrintClient.ajaxWithData @ packed_client.js?8f609640:7
OctoPrintClient.post @ packed_client.js?8f609640:7
OctoPrintClient.postJson @ packed_client.js?8f609640:7
OctoPrintSettingsClient.save @ packed_client.js?8f609640:57
SettingsViewModel.self.saveData @ packed_core.js?9ab43570:259
WizardViewModel.self.finishWizard @ packed_core.js?9ab43570:431
onFinish @ packed_core.js?9ab43570:430
finish @ packed_libs.js?ab739628:1209
dispatch @ packed_libs.js?ab739628:1
v.handle @ packed_libs.js?ab739628:1

cp2004, 2020-12-17:
@Mixdiver The issue has been fixed (it was not this plugin, but OctoPrint itself) for the next release (OctoPrint 1.6.0) but this might not be for a while.

In the meantime, please try testing out the release candidate as described in #16 as it does not have this 'new maintainer' dialog since it had been there for a while & was time to be removed.

RChadwick7, 2020-12-17:
I currently have this issue, and my Octoprint is now 100% unusable. I can SSH in to shutdown, but how do I disable this, or uninstall EEPROM?

cp2004, 2020-12-17:
@RChadwick7 go to the recovery page (`http://<your-ip>/recovery`), restart in safe mode & then you can either uninstall/disable the plugin, or install the [release candidate](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/16) for V3 of the plugin. Still needs a bit more testing before stable release, given it is a complete (and far better) rewrite it might be a short while before it goes into stable release. Hopefully before the end of the year though.

Mixdiver, 2020-12-18:
> @cp2004   go to the recovery page (`http://<your-ip>/recovery`), restart in safe mode 

That's what I've been looking for the the past week or so. Now I have access to OctoPrint to try updating SW.  THANKS!!

BSmith-66, 2020-12-18:
> @DreadPirateMiko @BSmith-66 @cohaolain @reggaemanu
> 
> Please test the release candidate of V3 of this plugin, it is available and this feature/fix is implemented.
> 
> Details can be found in the feedback ticket of how to install, update and report issues: #16

Tested and working now - thanks for the effort
(also just got my hands on a 4B)

cp2004, 2020-12-20:
3.0.0 has been released 🎉

closed 2020-12-20 by cp2004