cp2004/OctoPrint-EEPROM-Marlin #18: cant scroll through firmware information..

askm76, 2020-12-21:
#### What do you need help with?
i can,t scroll at all in the firmware info screen.. the rest show up fine... jsut installed octoprint and updated last thursday. then i got the update about 15 minutes ago...
#### Link to octoprint.log

cp2004, 2020-12-21:
Could you send a screenshot/video of the problem? It seems to scroll just fine for me here.

Maybe also a list of other plugins you have installed.

vovkas, 2020-12-22:
Prolem:
[image]

askm76, 2020-12-22:
Sorry. I rebooted both pi and PC. Cleared the cache and it's working now. Thanks for everything..Sent from my Galaxy

cp2004, 2020-12-22:
@vovkas please open a bug report filling out the *full* information, since that sounds like a different problem to this one (since OP wrote that the rest worked fine)

@askm76 good to hear it's fixed, thanks for reporting back.

closed 2020-12-22 by cp2004