cp2004/OctoPrint-EEPROM-Marlin #19: [Bug]Parse error

devildant, 2020-12-23:
#### Describe the bug

Parsing problem following the update

#### Steps To Reproduce

Load eeprom

#### Expected behavior

....

#### Logs

`octoprint.log`:
```
2020-12-23 22:53:10,363 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2020-12-23 22:53:10,417 - octoprint.plugins.eeprom_marlin - ERROR - Unrecognised EEPROM output line, could not parse
2020-12-23 22:53:10,419 - octoprint.plugins.eeprom_marlin - ERROR - Line: echo:  M149 C ; Units in Celsius
2020-12-23 22:53:10,550 - octoprint.plugins.eeprom_marlin - ERROR - Unrecognised EEPROM output line, could not parse
2020-12-23 22:53:10,551 - octoprint.plugins.eeprom_marlin - ERROR - Line: echo:  M603 L100.00 U120.00
2020-12-23 22:53:10,554 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI

```
#### Versions, system information

**Plugin** Version: 3.0.0

**OctoPrint**

Version: 1.5.2

Operating System running OctoPrint:

**Printer** model & firmware version: CR-10S with marlin 1.1.9

**Browser:**

**Additional context, screenshots, anything else you think might be useful**

devildant, 2020-12-23:
EEPROM FORMAT : 
```
Send: M503
Recv: echo:  G21    ; (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E415.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z10.00 E50.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X300 Y300 Z100 E5000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P2500.00 R500.00 T2500.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q50000 S0.00 T0.00 X11.00 Y10.00 Z0.20 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z4.00
Recv: echo:  G29 W I0 J0 Z-0.17350
Recv: echo:  G29 W I1 J0 Z-0.07850
Recv: echo:  G29 W I2 J0 Z-0.07150
Recv: echo:  G29 W I3 J0 Z-0.00400
Recv: echo:  G29 W I4 J0 Z-0.00750
Recv: echo:  G29 W I0 J1 Z-0.12750
Recv: echo:  G29 W I1 J1 Z-0.05150
Recv: echo:  G29 W I2 J1 Z-0.04900
Recv: echo:  G29 W I3 J1 Z0.01500
Recv: echo:  G29 W I4 J1 Z0.01750
Recv: echo:  G29 W I0 J2 Z-0.10750
Recv: echo:  G29 W I1 J2 Z-0.03150
Recv: echo:  G29 W I2 J2 Z0.00450
Recv: echo:  G29 W I3 J2 Z0.04100
Recv: echo:  G29 W I4 J2 Z0.06250
Recv: echo:  G29 W I0 J3 Z-0.09350
Recv: echo:  G29 W I1 J3 Z-0.01300
Recv: echo:  G29 W I2 J3 Z0.01550
Recv: echo:  G29 W I3 J3 Z0.05250
Recv: echo:  G29 W I4 J3 Z0.07100
Recv: echo:  G29 W I0 J4 Z-0.11150
Recv: echo:  G29 W I1 J4 Z-0.01150
Recv: echo:  G29 W I2 J4 Z0.00700
Recv: echo:  G29 W I3 J4 Z0.05650
Recv: echo:  G29 W I4 J4 Z0.07100
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H200 B60 F0
Recv: echo:  M145 S1 H240 B100 F0
Recv: echo:PID settings:
Recv: echo:  M301 P34.80 I3.10 D97.63
Recv: echo:  M304 P236.78 I38.54 D363.69
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-2.28
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L100.00 U120.00
```

cp2004, 2020-12-24:
Well, this is less of a bug and more of a convenient way of me adding more 'supported commands' to the plugin. I should probably change it from error to warning, since there is not really a 'problem' as such, just those commands are not recognized. Previously it would fail silently, now I made it log something.

I'll probably add M603 but not the M149 since it's not really that useful (imo) to change the temperature units.

devildant, 2020-12-24:
hello,
you're right, it's feature request :) (sorry)
yes M149 it's useless

cp2004, 2021-01-03:
Implemented for 3.0.1, brought forward due to the bug. Since this is a very small change, I managed to fit it in.

commit 4b17e80a (2021-01-03): :sparkles: Add M603 change filament

Closes #19
closed 2021-01-04 by cp2004