cp2004/OctoPrint-EEPROM-Marlin #20: [Bug]View broken (unusable) with touch ui

devildant, 2020-12-23:
#### Describe the bug
Hello, 
small problem with the touchui plugin and your plugin
[image]

#### Steps To Reproduce

install touch ui, use chrome with dev mode and swith to device view with touchui enable

#### Expected behavior

#### Logs

`octoprint.log`:

#### Versions, system information

**Plugin** Version: 3.0.0

**OctoPrint**

Version: 1.5.2

Operating System running OctoPrint: octopi 0.17.0

**Printer** model & firmware version: CR-10S marlin 1.1.9

**Browser:** chrome safari (ios)...

**Additional context, screenshots, anything else you think might be useful**

devildant, 2020-12-24:
feature reuest too :)

cp2004, 2020-12-24:
Interestingly, the table clipping off the side is actually a touchUI problem (IMO) since it applies the `width: 100mv` (meaning 100%). 

Overriding this, to set `width: auto; min-width: 100%` on the style improves this significantly:
[image]

For the icons, that is also touchUI being a little annoying in applying it's custom font to *everything*, which wipes out my custom FA5 icon set. I might be able to override this too, but it is not going to be clean lol:
[image]

A little extra margin can be applied to buttons to make them look a little nicer, maybe just the backup button.

Maybe you didn't want to know all that, but that was from my experimenting the browser console so I needed to keep track of it!

cp2004, 2021-01-04:
Similarly to #19, brought forward as it is a minor change that *might as well* go out with the bugfix release later today. It's the best I can do, to make it somewhat coordinated without re-writing the whole interface if touch UI is enabled.

commit 61ef473c (2021-01-04): :lipstick: Improvements to styles with touchUI

Closes #20

This is the best I can do, without re-writing the whole UI, or contributing to touchUI either to provide a different UI.
closed 2021-01-04 by cp2004