cp2004/OctoPrint-EEPROM-Marlin #21: Firmware Info - Firmware Name (with date and time) incorrectly formatted

kalle-1989, 2020-12-31:
**Describe the bug**

Hello, 

I have observed a formatting issue with the "Firmware Name" in the "Firmware Info" Tab.

This is the Firmware Name from the OctoPrint Log: "FIRMWARE_NAME:Marlin 2.0.7.2 (Dec 30 2020 01:08:32)"
[image]

but the EEPROM Editor Plugin shows the time "01:08:32" in two new lines incorrectly formatted.

[image].

#### Expected behavior
Firmware Name is displayed in a line with correctly formatted time.

**Versions, system information**
Plugin Version: 3.0.0

**OctoPrint**

Version: 1.5.2 (Python 3.7.3, OctoPi 0.17.0)

Operating System running OctoPrint:

Raspberry Pi 3 Model B Plus Rev 1.3

Printer model & firmware version: CR-10S with marlin 2.0.7.2

Browser:
Opera, Version: 73.0.3856.284
Firefox, Version: 84.0.1 (64-Bit)

cp2004, 2020-12-31:
It's not a bug I can fix (well, I could, but it would be more effort than it was worth) since it is not *actually* a bug in my code.

The name is parsed by core OctoPrint, to split out the name from extra info. It splits on colons currently, which obviously is wrong. We've discussed this already, and come up with a fixed way of parsing it. If you're interested, [discussion (`#dev-core`) on discord.](https://discord.com/channels/704958479194128507/705047010641838211/788136463732047932)

I'm not going to fix this within my plugin, and instead it will be fixed by OctoPrint 1.6.0, and work like magic with this plugin when that comes out. I could pull out the firmware parsing methods and do them myself, but I'd rather just fix the *actual* issue upstream.

kalle-1989, 2020-12-31:
Thanks for your fast reply. I will wait for OctoPrint 1.6.0.

closed 2020-12-31 by kalle-1989