cp2004/OctoPrint-EEPROM-Marlin #22: [Bug]

chriskoups, 2021-01-02:
Hello,

I recently installed this plugin on octoprint 1.5.2 for my CR-6 SE which was flushed with the latest RC5 firmware.
As usual, I try to set the z jerk to 0.5 instead of 0.3 and I noticed that the E jerk changed to 0.
Checking the command this EEPROM editor was sending:
M205 S0 T0.0 B20000.0 X10.0 Y10.0 Z0.5 E5.0 JNone
The last part of JNone is screwing the eeprom.
I assume is an easy fix somewhere in your scripts.
My machine does not report a J value by default, so I assume is 0.

cp2004, 2021-01-02:
Ooops! Missing a None check by the looks of it, in the command constructor:
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/blob/991339c85e7c6d2ca54f583196817a08d8387708/octoprint_eeprom_marlin/util.py#L15-L20

Since the data is stored as None in the server, as it is not reported. Will fix this and push out by the end of this weekend, fairly important problem. Thanks for reporting this!

cp2004, 2021-01-02:
Fixed for 3.0.1

commit fe0b2afd (2021-01-02): :bug: Additional None check for sending a command

Closes #22
closed 2021-01-04 by cp2004