cp2004/OctoPrint-EEPROM-Marlin #24: [Request] Bed Level mesh

KenLucke, 2021-01-04:
#### Is your feature request related to a problem? Please describe.
Please add the retrieval, display, editing, and storage for the bed level mesh. (accessible with M420 V)

Unfortunately, there's a bug somewhere in Marlin that loses the latest bed level mesh (and reverts to the previous one, how that happens, I haven't a clue) if you turn the printer controller off and on, forcing a new mesh to be generated for accuracy.  You can see this by doing a bed level, looking at it in the Bed Visualizer plugin, turning the machine off and on, and looking at it again in the Bed Visualizer.  I believe I recall @jneilliii discussing it somewhere in that plugin's issues, but I can't remember which thread.  I also know that the guy who compiles the fantastic Marlin versions (that I use with a few minor tweaks of my own) for the CR-10v2/3 on facebook (Nic Wilson) has described and discussed it as well.

#### Describe the solution you'd like
If the EEPROM Editor could retrieve, display and save (and allow editing if you have a .csv to enter), it would eliminate the need to re-do the bed level, because you could just load and store a backup with the proper mesh.

#### Describe alternatives you've considered
Doing the [in my case, 81 point] bed level every time the machine has to be cycled for some other reason, despite the bed level having not changed.

#### Additional context

cp2004, 2021-01-04:
You do like to challenge me... We'll see what happens. Might be able to ~~steal~~ borrow the code from Bed level visualiser that generates the table of values, and make it so it can be edited, since I'm not good enough at KO bindings to work that one out (yet).

Also more feasible now I have an ABL system, since I can actually test this myself. Would be an interesting one to work through, for sure.

jneilliii, 2021-01-04:
I don't really remember this one either Ken, but almost sounds like RESTORE_LEVELING_AFTER_G28 is not enabled in the firmware? @cp2004 my plugin does have an event hook your plugin could use to retrieve the mesh data....

KenLucke, 2021-01-04:
> You do like to challenge me... We'll see what happens. Might be able to ~steal~ borrow the code from Bed level visualiser that generates the table of values, and make it so it can be edited, since [...]

LOL, have to keep you on your toes!

KenLucke, 2021-01-04:
> I don't really remember this one either Ken, but almost sounds like RESTORE_LEVELING_AFTER_G28 is not enabled in the firmware? @cp2004 my plugin does have an event hook your plugin could use to retrieve the mesh data....

No, it's enabled, i double checked the firmware build.

KenLucke, 2021-01-04:
> > I don't really remember this one either Ken, but almost sounds like RESTORE_LEVELING_AFTER_G28 is not enabled in the firmware? @cp2004 my plugin does have an event hook your plugin could use to retrieve the mesh data....
> 
> No, it's enabled, i double checked the firmware build.

```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#if ANY(ABL_EZABL, ABL_BLTOUCH, ABL_NCSW, ABL_TOUCH_MI)
  #if ENABLED(ABL_UBL)
    #define AUTO_BED_LEVELING_UBL
  #elif ENABLED(ABL_BI)
    #define AUTO_BED_LEVELING_BILINEAR
  #endif
#else
  #define MESH_BED_LEVELING
#endif
/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28

/**
```

KenLucke, 2021-01-04:
I guess Nic hasn't done too much research into the bug, here's his response when I asked him to give me all the info he had on it:

> I have never looked deep into it,  I discovered first when making those hex lights where the first layer is super important, I found that following a power up,  using G29 or using M420 S1 in the print, did not do what was expected in that the nozzle was too close to the bed and not at its correct height by z-offset.  To resolve it,  all I do is a home,  and bed level from the menu,  and save it,   and from that point up until the next power up  the M420 or G29 works fine,   so it some bug that causes the 0+z-offset to be wrong,  unless a full bed level and save is done.

cp2004, 2021-02-23:
Some news on this one - there's a work-in-progress plugin for UBL mesh editing:

https://github.com/The-EG/OctoPrint-UBLMeshEdit/

Seen some demonstrations on the OctoPrint discord server, it looks very promising. If (when!) it makes it to the plugin repo, I don't think there's going to be much point in me implementing the same thing here.

KenLucke, 2021-02-23:
> Some news on this one - there's a work-in-progress plugin for UBL mesh editing:
>
> https://github.com/The-EG/OctoPrint-UBLMeshEdit/
>
> Seen some demonstrations on the OctoPrint discord server, it looks [...]

Thanks for the update!

cp2004, 2021-03-13:
Going to close this, since the mentioned plugin seems to be going quite well having launched on the repository.

closed 2021-03-13 by cp2004