cp2004/OctoPrint-EEPROM-Marlin #25: [Bug] Make Control Active (tiny, tiny bug)

KenLucke, 2021-01-04:
#### Describe the bug
When the Named Backup dialog opens, despite the obvious intent to enter a name, and a single control (TextEdit box), it still requires a discrete click in to the TextEdit box to activate it for typing.

#### Steps To Reproduce
Attempt to save a backup with "Named Backups" enabled

#### Expected behavior
The text edit box should automatically become active for typing upon the dialog opening.

#### Logs
Not really relevant to this one.

#### Versions, system information
Marlin EEPROM Editor: 3.0.1 

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3927818240
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020

cp2004, 2021-01-04:
Implemented via the above commit, ready for whatever release comes next. Still had everything open from the release earlier today, so you get it implemented both fast and slow. Fast, since it is 20 mins since you posted this, slow because it will have to wait for the next release. Unless... I release this one now since the rollout of 3.0.1 will not have gone far. Got any more tiny things waiting?

KenLucke, 2021-01-04:
> Implemented via the above commit, ready for whatever release comes next. Still had everything open from the release earlier today, so you get it implemented both fast and slow. Fast, since it is 20 [...]

Now THAT's service :).

Don't mind waiting for next release at all.  As I said, it was a tiny, tiny thing.

Did you get to work on the Runout Sensor status thing we talked about in #16?

cp2004, 2021-01-04:
No, I promptly forgot about it

KenLucke, 2021-01-04:
> No, I promptly forgot about it

Shame, shame.....   :)

closed 2021-03-21 by cp2004

cp2004, 2021-03-21:
3.0.2 was released just now, with this in it.

commit c2761446 (2021-01-04): :lipstick: Auto-focus input when creating a named backup

Closes #25