cp2004/OctoPrint-EEPROM-Marlin #26: Delta Configuration Not Matching Current Marlin 2.0.x[Bug]

Firworksyt, 2021-01-05:
#### Describe the bug
The Delta settings page in the plugin does not match the response from the M665 Gcode command on the current version of Marlin.

#### Steps To Reproduce

On printer running Marlin 2.0.7.2
1. Issue M503 in terminal
2. Note format of M665
3. Go to Delta Settings in plugin
4. Note incorrect calibration radius which contains the value of A rod trim.

#### Expected behavior
Delta settings screen in plugin matches definition of M665 command and supports view/edit/backup/restore of diagonal rod length adjustments.

#### Logs

`octoprint.log`:
2021-01-05 04:19:21,676 - octoprint.startup - INFO - ******************************************************************************
2021-01-05 04:19:21,677 - octoprint.startup - INFO - Starting OctoPrint 1.5.2
2021-01-05 04:19:21,678 - octoprint.startup - INFO - ******************************************************************************
2021-01-05 04:19:22,090 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-01-05 04:19:22,148 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-01-05 04:19:22,148 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-01-05 04:19:22,149 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-01-05 04:19:22,564 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2021-01-05 04:19:22,584 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-01-05 04:19:23,402 - octoprint.plugin.core - INFO - Plugin Marlin Flasher (0.1.4) is disabled.
2021-01-05 04:19:23,553 - octoprint.plugin.core - INFO - Plugin Octolapse (0.3.4) is disabled.
2021-01-05 04:19:23,634 - octoprint.plugin.core - INFO - Plugin Delta-Micro-Calibrator (1.0.4) is disabled.
2021-01-05 04:19:23,818 - octoprint.startup - INFO - Added settings overlay from plugin dashboard
2021-01-05 04:19:23,827 - octoprint.plugin.core - INFO - Found 30 plugin(s) providing 27 mixin implementations, 60 hook handlers
2021-01-05 04:19:23,919 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-01-05 04:19:23,934 - octoprint.server - INFO - Intermediary server started
2021-01-05 04:19:23,951 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2021-01-05 04:19:24,540 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2021-01-05 04:19:24,543 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: u"Need(method='role', value=u'plugin_action_command_notification_show')")
2021-01-05 04:19:24,544 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: u"Need(method='role', value=u'plugin_action_command_notification_clear')")
2021-01-05 04:19:24,545 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2021-01-05 04:19:24,546 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2021-01-05 04:19:24,547 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plugin_announcements_read')")
2021-01-05 04:19:24,548 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2021-01-05 04:19:24,549 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2021-01-05 04:19:24,549 - octoprint.server - INFO - Added new permission from plugin dashboard: PLUGIN_DASHBOARD_ADMIN (needs: u"Need(method='role', value=u'plugin_dashboard_admin')")
2021-01-05 04:19:24,550 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_firmware_check_display')")
2021-01-05 04:19:24,552 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2021-01-05 04:19:24,552 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2021-01-05 04:19:24,553 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2021-01-05 04:19:24,554 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'plugin_pluginmanager_manage')")
2021-01-05 04:19:24,555 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: u"Need(method='role', value=u'plugin_softwareupdate_check')")
2021-01-05 04:19:24,556 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: u"Need(method='role', value=u'plugin_softwareupdate_update')")
2021-01-05 04:19:24,557 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: u"Need(method='role', value=u'plugin_softwareupdate_configure')")
2021-01-05 04:19:24,801 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-01-05 04:19:24,838 - octoprint.plugins.excluderegion - INFO - Logging mode set to 'octoprint'
2021-01-05 04:19:24,840 - octoprint.plugins.excluderegion - INFO - Setting update detected: g90InfluencesExtruder=False, clearRegionsAfterPrintFinishes=False, mayShrinkRegionsWhilePrinting=False, loggingMode=octoprint, enteringExcludedRegionGcode=None, exitingExcludedRegionGcode=None, extendedExcludeGcodes={'M73': {"gcode": "M73", "type": "ExcludedGcode", "mode": "merge", "description": "Suppress progress updates while excluding and output the most recent progress values encountered when exiting the excluded area"}, 'M117': {"gcode": "M117", "type": "ExcludedGcode", "mode": "last", "description": "Suppress display messages while excluding and output the last message encountered when exiting the excluded area"}, 'G4': {"gcode": "G4", "type": "ExcludedGcode", "mode": "exclude", "description": "Ignore all dwell commands in an excluded area to reduce delays while excluding"}, 'M205': {"gcode": "M205", "type": "ExcludedGcode", "mode": "merge", "description": "Record advanced setting changes while excluding and apply the most recent values in a single command after exiting the excluded area"}, 'M204': {"gcode": "M204", "type": "ExcludedGcode", "mode": "merge", "description": "Record default acceleration changes while excluding and apply the most recent values in a single command after exiting the excluded area"}}, atCommandActions={'ExcludeRegion': [{"action": "enable_exclusion", "type": "AtCommandAction", "command": "ExcludeRegion", "parameterPattern": "^\\s*(enable|on)(\\s|$)", "description": "Default action to enable exclusion"}, {"action": "disable_exclusion", "type": "AtCommandAction", "command": "ExcludeRegion", "parameterPattern": "^\\s*(disable|off)(\\s|$)", "description": "Default action to disable exclusion"}]}
2021-01-05 04:19:24,865 - octoprint.plugins.eeprom_marlin - INFO - EEPROM Data initialised
2021-01-05 04:19:24,867 - octoprint.plugins.eeprom_marlin - INFO - Backup metadata initialised
[... cut ...]
Recv: echo:; Delta settings: L<diagonal rod> R<radius> H<height> S<segments per sec> XYZ<tower angle trim> ABC<rod trim>
Recv: echo:  M665 L267.0000 R133.5951 H270.6000 S80.0000 X0.2631 Y-0.0122 Z-0.2509 A0.0000 B0.0000 C0.0000

I've edited the current plugin already to successfully support this but I cannot figure out how to actually setup a proper development environment as per the contribution guidelines to make a commit and pull request for this issue. The above log and behavior report however is with a clean fresh install of the plugin.

cp2004, 2021-01-05:
So I'm reading the documentation for marlin (https://marlinfw.org/docs/gcode/M665.html) about this code, and it lists the B parameter as the calibration radius. As far as I can tell, the plugin should be using this B parameter in this field. Could you explain a little more about which parameter you expected to be under 'Calibration radius' label? 

I don't have a delta machine, so have to go from the docs to create the plugin, or by someone else telling me otherwise. Might be able to read the source of the command, or ask in the Marlin discord to see if they know more info about the docs being different.

Firworksyt, 2021-01-05:
I'll see if I can find the thread on the Marlin Github but they've mentioned that documentation is out of date and haven't gotten around to updating it still.

cp2004, 2021-01-05:
@Firworksyt You mentioned you adjusted the plugin to work for you, if you can upload the files here I would be able to add them (with you as the co-author). I'll also ask on the Marlin discord at some point, unless maybe reading the source would be easier 😄.

Firworksyt, 2021-01-07:
Sure, I'll clean it up some and test it out a bit more and then attach it here. I made the initial edit to see if I could figure it out and then came here when it worked.

I looked at the current Marlin source and here's where they're declaring the data structure to store delta configurations:

https://github.com/MarlinFirmware/Marlin/blob/f74015b4e57677cd033e116dab2331478edbe2ec/Marlin/src/module/settings.cpp#L266-L282

Here's where they're parsing a received M665 message:

https://github.com/MarlinFirmware/Marlin/blob/f74015b4e57677cd033e116dab2331478edbe2ec/Marlin/src/gcode/calibrate/M665.cpp#L30-L59

Here's where they're sending out a M665 as part of the settings report in response to something like an M503:

https://github.com/MarlinFirmware/Marlin/blob/f74015b4e57677cd033e116dab2331478edbe2ec/Marlin/src/module/settings.cpp#L3207-L3230

All of them show this format with ABC for rod length so I think the documentation is just wrong.

One question I had while looking at these though is how you want to handle SCARA printers. I know they're even less common than Deltas but it looks like Marlin has overlaid the M665 message to either be a Delta config or SCARA config. If someone with a SCARA printer tries the plugin it'll probably behave very strangely. The start of the M665 response does say either "SCARA settings:" or "Delta settings:" so you could probably key off of that.

cp2004, 2021-01-07:
Yeah, SCARA is not handled at all. Could probably be done, with a flag in it's data entry. That's not actually something I decided on, since I adopted this plugin back in October. Of course, with V3 I re-wrote most of it but the concept of what commands etc. was still there from the old version. Now I've got it into a decent framework, we can think about things like this. Interestingly, in the 5 years this plugin has been around, no one has mentioned SCARA support in a single issue, out of [over 5K tracked installs](https://cp2004.github.io/OctoPluginStats). I'll see, if it is simple enough to set a flag (based on the response you mention) to change things, otherwise it is very low priority.

Thanks for the links, saves me some time clicking and trying to comprehend the Marlin structure for a while 🙂.

cp2004, 2021-03-28:
Alright, I think I got this one. I'll let you know when there's a pre-release available to test, since I don't have a delta I hope I've read all the sources you posted correctly.

cp2004, 2021-05-29:
There's a release candidate available for you to test this now: [3.1.0rc1](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.1.0rc1). Feedback ticket #45 for any comments on the release. Stable release approx. 1 week depending on how the testing goes.

commit 15922762 (2021-03-28): :bug: Fix delta settings for Marlin 2.0.x

Closes #26
closed 2021-06-07 by cp2004