cp2004/OctoPrint-EEPROM-Marlin #28: [Request]  Another data point to check : MeatPack

KenLucke, 2021-02-07:
This looks as if it will get supported in a variety of firmware branches (I know that the guy doing my firmware, and that for several Creality models is already testing it).  Can you add a check for it?

https://plugins.octoprint.org/plugins/meatpack/

cp2004, 2021-02-07:
I would imagine it would be reported in the capability report (M115), in which case it should just show up if there is a branch of firmware that has this implemented - previous ones don't say 'i don't support it' because they don't know it exists.

If it doesn't show up in the capability report... Well then that's more difficult.

cp2004, 2021-02-23:
@KenLucke Can you confirm that on firmware with the MeatPack feature existing this shows up in the cap report?

KenLucke, 2021-02-23:
Yes, it does.

[image]

KenLucke, 2021-02-23:
As I opened this, and it was [automatically] resolved, I might as well close it.

closed 2021-02-23 by KenLucke

cp2004, 2021-02-23:
Thanks for confirming 🙂