cp2004/OctoPrint-EEPROM-Marlin #29: [Request] Disable ability to write to EEPROM for some users, eg. members of Read-only Access group

dat-atr-ash, 2021-02-10:
#### Is your feature request related to a problem? Please describe.

It seems that any user with access to octoprints web interface can use the plugin in to write to EEPROM, including members of the Read-only Access group which makes it problematic. 

#### Describe the solution you'd like

It would be great if the permission management system already present in octoprint could be used to disable access for chosen users and/or groups of users.

#### Describe alternatives you've considered

Not really since the only alternative would be uninstalling. Or maybe enabling just when needed, but...

#### Additional context

I wanted to set up an account for another household member that should not be allowed access to the printer itself.

cp2004, 2021-03-27:
With https://github.com/cp2004/OctoPrint-EEPROM-Marlin/commit/a9197dbdf5ba756b58775682158d84d3e3483352 this has been implemented, ready for 3.1.0.

cp2004, 2021-05-29:
There's a release candidate available for testing this now: [3.1.0rc1](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.1.0rc1). Feedback ticket #45 for any comments on the release. Stable release approx. 1 week depending on how the testing goes.

cp2004, 2021-06-07:
3.1.0 has been released

closed 2021-06-07 by cp2004