cp2004/OctoPrint-EEPROM-Marlin #30: [Request] Add TMC Support

rodmagno, 2021-02-11:
It would be nice to have TMC drivers info and edit the values

https://marlinfw.org/docs/gcode/M122.html
https://marlinfw.org/docs/gcode/M569.html
https://marlinfw.org/docs/gcode/M906.html
https://marlinfw.org/docs/gcode/M913.html

cp2004, 2021-02-11:
I will see what can be done. It will need testing from you when it is done - I have to extend the virtual printer for each of the commands I want to test, my printer does not have this stuff, and it doesn't always substitute for the real thing.

rodmagno, 2021-02-12:
Sure, whenever you have a snapshot, let me know and i will test it

Hobby4life, 2021-02-23:
Would indeed be a great implementation :-)

cp2004, 2021-05-27:
Could someone here send me the M503 response from your printer? I need to implement and test the parser for these commands, some of them look more complex than the regular ones. Thank you in advance!

cp2004, 2021-05-29:
@rodmagno or @Hobby4life - I have a development build available for you to test, if you want. It has support for M906 and M913 commands. The others are more complex and will have to wait until I have a more flexible data structure involved. Additionally, it only has support for configuring this globally, there is no per-motor control implemented for the same reason.

If you want to try the build, install using the URL:
```
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/6fbf51ee35e1a0e6b678d6b6f2a2b6b57f845b0c.zip
```
Let me know what you think.

cp2004, 2021-05-29:
There's also a release candidate available for testing this now: [3.1.0rc1](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.1.0rc1). Feedback ticket #45 for any comments on the release. Stable release approx. 1 week depending on how the testing goes.

cp2004, 2021-06-07:
3.1.0 has been released

closed 2021-06-07 by cp2004