cp2004/OctoPrint-EEPROM-Marlin #31: [Bug] Import unknown module

sebastian-be, 2021-02-14:
Hi

Running v3.0.1 on Octoprint 1.3.12, with python 2.7, get an error on startup. See logs below.
Issue is in octoprint_eeprom_marlin/util.py, introduced in commit 4e9690c0477e078879eb3a7fe899ca6ba8e76a4c.

Fix:
<<<
from octoprint.vendor.awesome_slugify import Slugify
>>>
try:
    # OctoPrint>=1.4.1
    from octoprint.vendor.awesome_slugify import Slugify
except ImportError:
    # OctoPrint<=1.4.0
    from slugify import Slugify

Logs:
2021-02-14 13:18:47,254 - octoprint.plugin.core - ERROR - Error loading plugin eeprom_marlin
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin/__init__.py", line 17, in <module>
    from octoprint_eeprom_marlin import (
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin/api.py", line 13, in <module>
    from octoprint_eeprom_marlin import util
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin/util.py", line 50, in <module>
    from octoprint.vendor.awesome_slugify import Slugify
ImportError: No module named awesome_slugify

cp2004, 2021-02-14:
I can probably adjust that at some point.

However, you really should update to a newer version of OctoPrint. You're using 1.3.12, which is close to a year and a half out of date.

sebastian-be, 2021-02-14:
I've already fixed it, see original comment (edited it now to make it more obvious).
If you mean by "out of date" -> "released" then sure...
https://octoprint.org/blog/2019/10/22/new-release-1.3.12/

cp2004, 2021-03-11:
This has been adjusted ready for 3.1.0

commit 843586a7 (2021-03-11): :bug: Fix plugin import under OctoPrint < 1.4.1

Closes #31
closed 2021-03-12 by sebastian-be