cp2004/OctoPrint-EEPROM-Marlin #32: [Bug] Firmware Info Name parsing

LMS0815, 2021-02-18:
#### Describe the bug

Firmware Info has a phasing or display bug

#### Steps To Reproduce

See image:
[image]

Firmware is: FIRMWARE_NAME:Marlin 2.0.6_DW7 TM3D SSMDBU (Sep 25 2020 11:41:48) SOURCE_CODE_URL:https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0 PROTOCOL_VERSION:1.0 MACHINE_TYPE:TM3D CR-X EXTRUDER_COUNT:2 UUID:xxxxxxxxx

#### Expected behavior

**Firmware Name:** Marlin 2.9.6_DW7 TM3D SSMDBU (Sep 25 2020 11:41:48)

cp2004, 2021-02-18:
Duplicate of #21, it is not a problem with my plugin but actually OctoPrint, which gets upset with the time.

We're working on it, but it is being a pain trying to figure out the parsing of it. I have opened https://github.com/OctoPrint/OctoPrint/pull/3916 and we've discussed on Discord how to get this resolved.

Once a fix is in place in OctoPrint, then the firmware name in this plugin will naturally be fixed as well without any additional changes to this plugin.

closed 2021-02-18 by cp2004

LMS0815, 2021-02-19:
Thank you. I haven't seen the Duplicate.

cp2004, 2021-02-23:
This has now been fixed upstream in OctoPrint, so will work in 1.6.0+