cp2004/OctoPrint-EEPROM-Marlin #33: [Bug] Linear advance not showing the right figures

LMS0815, 2021-02-18:
#### Describe the bug

K factor is: 0

[image]

#### Steps To Reproduce

```
M900
echo:Advance K 0:1.50 1:0.00

```
#### Expected behavior

Should report 1,5

cp2004, 2021-02-18:
Please upload the logs as set out in the [bug report template](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/new?assignees=&labels=&template=bug_report.md&title=%5BBug%5D)

cp2004, 2021-02-18:
Ok, I think I see the issue - you have multiple extruders? What's with this `K 0:1.5.0 1:0.00` response? I have never seen this before.

LMS0815, 2021-02-19:
Yes multiple (two) extrudes (but only one hotend). 
The result I shared is provided when sending M900 on the Terminal window.

LMS0815, 2021-02-20:
I assume the issue is related to all extruder related values, as your plugin is obviously not displaying multiple lines if there is more than one extruder (although you already know the count)..

See: M900 or M603 below...

```
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y5.00 Z0.30 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Unified Bed Leveling:
echo:  M420 S1 Z10.00

Unified Bed Leveling System v1.01 active

echo:; Active Mesh Slot: 0
echo:; EEPROM can hold 21 meshes.

echo:; PID settings:
echo:  M301 P27.72 I2.73 D70.43
echo:  M304 P189.89 I37.34 D643.72
echo:; Power-Loss Recovery:
echo:  M413 S0
echo:; Z-Probe Offset (mm):
echo:  M851 X-12.50 Y-38.00 Z-1.70
echo:; Linear Advance:
echo:  M900 T0 K1.50
echo:  M900 T1 K1.50
echo:; Filament load/unload lengths:
echo:  M603 T0 L450.00 U430.00
echo:  M603 T1 L450.00 U430.00
echo:; Tool-changing:
echo:  M217 S80.00 B0.00 E2.00 P276.00 R3000.00 U1500.00 F255 G10 A0 L0.00 W1.00 X315.00 Y10.00 Z2.00
```

cp2004, 2021-05-29:
I'm not able to reproduce this issue on my development build, I've built a debug feature & thrown the M503 response above at it - it is parsing the command OK. Multi-extruder support is not implemented, but it should be able to cope with finding the value. Multi extruder support is tracked in #40.
[image]

If you want to try this development build, you can install it using the URL:
```
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/5eba1a9f08f9eab03aeaf74553dbba95bc2f71ed.zip
```

cp2004, 2021-06-07:
Since I couldn't reproduce and haven't heard anything, I'll close this. Feel free to request a reopen or open a new issue, I'm keeping an eye on this one with some unit tests so hopefully it doesn't break.

closed 2021-06-07 by cp2004