cp2004/OctoPrint-EEPROM-Marlin #34: [Bug]

b-morgan, 2021-03-15:
The plugin manager entry for this plugin does not have a Homepage link like most of the other plugins (see attached screenshot)

1. Open the Plugin Manager
2. Scroll down to Marlin EEPROM Editor
3. See error

#### Expected behavior

I expected to see a Homepage link

#### Logs

`octoprint.log`: https://www.mediafire.com/file/sffpqm9jqq7224d/octoprint.log/file

#### Versions, system information

**Plugin** Version: 3.0.1

**OctoPrint**

Version: 1.5.3

Operating System running OctoPrint: OctoPi 0.18.0

Printer: LulzBot TAZ 6, Marlin bugfix-2.0.x (Sep 13 2020 12: 13) 

Browser: Firefox 86.0 (64-bit)

[image]

cp2004, 2021-03-15:
Hmm, the website is defined in setup.py. None of the metadata that is in setup.py shows up for some reason, only the overridden stuff. Wonder if there is a name mismatch or something. Either way, if I am lazy I will just override and define it again, or *actually* fix the issue, for 3.1.0.

cp2004, 2021-03-15:
Having just commented that, I can actually see the metadata in my dev install, but not on my other printer's install 🤔

cp2004, 2021-05-29:
There's a release candidate available for testing this plugin now: [3.1.0rc1](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.1.0rc1). Feedback ticket #45 for any comments on the release. Stable release approx. 1 week depending on how the testing goes.

commit a475aab7 (2021-05-25): :bug: Add link to homepage inside plugin

Closes #34
closed 2021-06-07 by cp2004