cp2004/OctoPrint-EEPROM-Marlin #36: [Request] Please add some missing settings

Taomyn, 2021-03-17:
#### Is your feature request related to a problem? Please describe.

Currently the plug-in reports the following are unknown to it:

```
2021-03-17 15:55:37,274 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2021-03-17 15:55:37,283 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,284 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M149 C ; Units in Celsius

2021-03-17 15:55:37,326 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,327 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M413 S1

2021-03-17 15:55:37,331 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,332 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 X580 Y580 Z580

2021-03-17 15:55:37,333 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,333 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 T0 E650

2021-03-17 15:55:37,336 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,336 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 X Y Z

2021-03-17 15:55:37,337 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,338 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 T0 E

2021-03-17 15:55:37,341 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-17 15:55:37,342 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M412 S1 D25.0000
```

#### Describe the solution you'd like

Please support them

#### Additional context

Ender 3 Pro running latest Marlin bugfix-2.0x, but I do not believe any of these codes are new.

cp2004, 2021-03-17:
Hey @Taomyn!

This plugin doesn't support anything close to everything. When I started work on it, I made a list of the commands I could find, which reached over 50. I have implemented somewhere around 20-30 of them.

Please describe *why* you want these commands added. It is not a simple process, of *just add them* so it is good to have some reasoning behind them. I'm  not working to add them 'because they are not added', please tell me why you want each command to be editable from the UI.

cp2004, 2021-03-25:
@Taomyn Friendly ping, since I am still waiting for a response here

Taomyn, 2021-03-26:
Pointless as a backup mechanism if you're not going to support all possible values that can be stored in the EEPROM

closed 2021-03-26 by Taomyn

cp2004, 2021-03-26:
I'm up for adding in new data, but I can't justify working for free to implement something that would not be worthwhile. If you want to implement and test them, I would be more than happy to merge a PR.