cp2004/OctoPrint-EEPROM-Marlin #37: EEPROM Settings Not Available

GriffinWalmsley, 2021-03-22:
Communication with EEPROM

I am using OctoPrint with the SKR Mini e3 v2 and I have the Marlin EEPROM editor installed. I can view some of the current settings but only the firmware info tab. I also have not been able to make any adjustments. I have set the basic settings:

#define EEPROM_SETTINGS
#define DISABLE_M503
#define EEPROM_CHITCHAT

I don't understand why the rest of the settings do not appear. Am I doing something wrong?

I have attached a picture of the interface that I am seeing.

[image]

cp2004, 2021-03-22:
Please press the load button, then copy the contents of the terminal to somewhere like https://gist.github.com and link the file.

[image]

[image]

cp2004, 2021-03-22:
Actually, I look again - `#define DISABLE_M503` must be **uncommented**. The plugin uses M503 to get the data out of the printer.

cp2004, 2021-03-27:
@GriffinWalmsley friendly ping, did the above make sense?

cp2004, 2021-04-13:
Going to assume this is sorted, closing the issue. If you have further questions please get back in contact.

closed 2021-04-13 by cp2004