cp2004/OctoPrint-EEPROM-Marlin #40: Multiple Extruders

dtylerb, 2021-04-11:
I have two different extruders with different e step values. In the "steps" section it only shows me the esteps for the second extruder (tool 1) and not the primary (tool 0).

This is a printer that I've recently customized myself, so there's a chance I did something wrong in the firmware, but I don't think so at the moment.

cp2004, 2021-04-12:
The plugin currently does not handle multiple extruders - I'm reworking the data structure to allow for this, but not sure of support for multiple extruders will make it in to the next version. I have no hardware to test multi-extruder stuff with, so have to reverse engineer all of it to work in the virtual printer.

cp2004, 2021-05-23:
@dtylerb, could you send across the full output of your M503 command? No plans to implement multi-extruder support in the next release, but I want to know what I *would* have to work with, to make some design decisions. Thanks 🙂.

dtylerb, 2021-05-24:
Here you go. Copied and pasted from the terminal output in Octoprint.

[M503.txt](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/6529149/M503.txt)

cp2004, 2021-05-24:
Thank you! Very helpful.

cp2004, 2022-12-28:
I have added support for multiple extruders in the EEPROM editor, to be released as version 3.2.0. There may be some release candidates before then, so if anyone are interested in trying it early (no guarantees as to exactly when it will drop), switch to the 'Release Candidate' option in the plugin's settings.

[image]

cp2004, 2022-12-29:
Now available as 3.3.0rc1.

closed 2023-01-03 by cp2004