cp2004/OctoPrint-EEPROM-Marlin #41: [Request] M78 support

KenLucke, 2021-04-18:
#### Is your feature request related to a problem? Please describe.

Could you please add M78 support to the Advanced Panel, so that printer stats show from the last time a "load" is done?  Not exactly EEPROM, as clearing the EEPROM doesn't reset it, but informational nonetheless

Send: M78
Recv: Stats: Prints: 1683, Finished: 1489, Failed: 194
Recv: Stats: Total time: 153d 23h 16m 42s, Longest job: 14d 7h 43m 9s
Recv: Stats: Filament used: 29846.78m

cp2004, 2021-04-25:
Not an editable thing either? Would probably make more sense on the info tab, if any of them

KenLucke, 2021-04-25:
Info tab would be fine :)

cp2004, 2021-11-13:
It's been a while! I had some time to implement this today, so I did!

[image]

Not sure when it will release, but it will be in 3.2.0.

KenLucke, 2021-11-13:
Awesome, thank you!

cp2004, 2021-11-14:
There's a release candidate build available if you want to test it - switch to 'Release Candidate' channel for this plugin in the software update settings in OctoPrint.

Release notes: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1
Feedback ticket: #56

KenLucke, 2021-11-14:
Just tried the pre-release.  It shows the "Printer Stats" (M78)" line, but nothing else.  

[image]

I thought it might be a conflict with UI_Customizer, but disabling that gives the same result

[image]

KenLucke, 2021-11-14:
Oh, M78, even when went by terminal, is not responding.

commit 3bc649d1 (2021-11-13): :sparkles: Report M78 statistics

Closes #41
closed 2021-12-31 by cp2004