cp2004/OctoPrint-EEPROM-Marlin #42: EEPROM Editor's 'Load/Save/Reset' are available to users with "Read-only access"

ronkinoz, 2021-04-29:
#### Describe the bug

EEPROM Editor's 'Load/Save/Reset' are available to users with "Read-only access".

#### Steps To Reproduce

1. Be logged in as an Admin user
2. Go to Access Control/Groups
3. Edit Guest user
4. Enable Subgroups/Read-only Access for Guest User
5. Confirm, Save
6. Logout
7. As Guest,  go to EEPROM Editor tab and trash the EEPROM contents. ;-)
 
#### Expected behavior

Save/Reset should not be available to users with "Read-only Access".

#### Logs

`octoprint.log`: https://pastebin.com/MR3tadfU

#### Versions, system information

**Plugin** Version: 3.0.2

**OctoPrint**

Version: 1.6.0

Operating System running OctoPrint: OctoPi 0.17.0

**Printer** model & firmware version: Lulzbot Mini, Firmware Version 1.1.9.34

**Browser:** Firefox 88.0

**Additional context, screenshots, anything else you think might be useful**

ronkinoz, 2021-04-29:
Doh - just realized this is a duplicate of #29

cp2004, 2021-04-29:
Fixed already in development, I am too busy to finish everything in-progress I have locally but I will get to it at some point. If you are concerned about security here and want it fixed, the commit is available on the `devel` branch.

a9197dbdf5ba756b58775682158d84d3e3483352 / install with `https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/refs/heads/devel.zip`

closed 2021-04-30 by cp2004