cp2004/OctoPrint-EEPROM-Marlin #43: Missing options in editor

nsarzyns, 2021-05-25:
#### What do you need help with?

When I select the the EEPROM Editor tab, I'm missing the Steps and Linear Advance; the latter might be not having Linear Advance enable in the firmware....I need to check on that but most definitely the Steps menu is missing.

#### Link to octoprint.log

https://pastebin.com/BJYmRU4D

cp2004, 2021-05-26:
Hmm, I'm not seeing those commands spat out in the log because it couldn't parse them. So there's two possibilities:
1. The steps aren't being reported from the M503 (Steps are the M92 command)
2. There's a bug rendering it in the UI somewhere.

Could you send M503 and paste the entire output here? If you do find M92 in there, to rule out a server side issue the next thing would be to check what is returned by the endpoint `/api/plugin/eeprom_marlin` - just open that in your browser and find the `steps` section.

nsarzyns, 2021-05-27:
M503 does report M92 through the octoprint terminal. 

```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.28 Z396.03 E98.41
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E60.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 W I0 J0 Z-0.51000
Recv: echo:  G29 W I1 J0 Z-0.54750
Recv: echo:  G29 W I2 J0 Z-0.53000
Recv: echo:  G29 W I3 J0 Z-0.53250
Recv: echo:  G29 W I4 J0 Z-0.56000
Recv: echo:  G29 W I0 J1 Z-0.57250
Recv: echo:  G29 W I1 J1 Z-0.61000
Recv: echo:  G29 W I2 J1 Z-0.59000
Recv: echo:  G29 W I3 J1 Z-0.60750
Recv: echo:  G29 W I4 J1 Z-0.64250
Recv: echo:  G29 W I0 J2 Z-0.59500
Recv: echo:  G29 W I1 J2 Z-0.61750
Recv: echo:  G29 W I2 J2 Z-0.60250
Recv: echo:  G29 W I3 J2 Z-0.62750
Recv: echo:  G29 W I4 J2 Z-0.65500
Recv: echo:  G29 W I0 J3 Z-0.58750
Recv: echo:  G29 W I1 J3 Z-0.62500
Recv: echo:  G29 W I2 J3 Z-0.61750
Recv: echo:  G29 W I3 J3 Z-0.63000
Recv: echo:  G29 W I4 J3 Z-0.64250
Recv: echo:  G29 W I0 J4 Z-0.54500
Recv: echo:  G29 W I1 J4 Z-0.60500
Recv: echo:  G29 W I2 J4 Z-0.60250
Recv: echo:  G29 W I3 J4 Z-0.61750
Recv: echo:  G29 W I4 J4 Z-0.64000
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H185.00 B45.00 F255
Recv: echo:  M145 S1 H240.00 B110.00 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P28.72 I2.82 D73.10
Recv: echo:  M304 P92.30 I17.71 D320.68
Recv: ; Controller Fan
Recv: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-39.80 Y-8.50 Z-1.71
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L350.00 U400.00
Recv: ok
```

The steps section returned by the end point is `"steps":{"command":"M92","params":{"E":null,"X":null,"Y":null,"Z":null}}},`

nsarzyns, 2021-05-27:
Just realized the M92 command returned through the terminal has the extruder steps at the end while the endpoint has it at the beginning. Is this the root of the issue?

cp2004, 2021-05-28:
OK, so that confirms that you're having an issue with the server side parser rather than the UI, thanks. The only thing different about the M92 command is the number of spaces at the start. As far as I knew, the parser did not care about the number of spaces at the front, but I will throw that M503 response at it to see what happens on my side. The order of the parameters does not matter here.

cp2004, 2021-05-29:
I'm not able to reproduce the issue, I've created a debug feature and it seems to parse the values just fine:
[image]

**However**, this is on my development build and there's a a few things that have changed here so it may either be fixed or some kind of conflict somewhere on your install.

If you want, you can try the development build I'm testing using this URL:
```
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/5eba1a9f08f9eab03aeaf74553dbba95bc2f71ed.zip
```

nsarzyns, 2021-05-29:
I did try the dev build supplied and the steps menu was still missing. I ended up going through my plugins and disabling them all and enabling one by one to see if a conflict arose. I did find the plugin which was causing a conflict, it's the Calibration-Companion (https://github.com/iFrostizz/OctoPrint-CalibrationCompanion). I never used it and installed it on a whim to see what it was since it was new.

Thanks for your help too!

cp2004, 2021-05-30:
Ah right, it looks like that plugin is looking out for the values for steps as well, maybe it swallows the communication somehow and doesn't allow this plugin to receive that line as well. If I get a chance I'll poke it and let the author know.

closed 2021-05-30 by cp2004

cp2004, 2021-06-03:
Should be fixed with https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/pull/19