cp2004/OctoPrint-EEPROM-Marlin #45: Feedback on the 3.1.0rc1 release candidate

cp2004, 2021-05-29:
It's here, 3.1.0rc1 Release Candidate!

Be sure to check out the release candidate [release notes here](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.1.0rc1) for details of all the changes.

Please test this release, and let me know what you think. If all goes well this will be released at some point next week, so feedback is important to make sure that everything is good! An 'All is working well' is great feedback, since it lets me know people *are* actually testing this release, even if there are no issues to report.

**If you find a bug** please use the bug report template and send across the logs, so that I can get it fixed as quickly as possible.

KenLucke, 2021-05-29:
Everything looks good, Charlie, except that when first opened after an OP reload, the info list is alphabetical, but when you "load", it scrambles the order.

cp2004, 2021-05-30:
It is two different methods of getting the same thing as well, the initial load is provided by the API, but then when you hit the 'load' button the data is sent via the websocket. The 'scrambled' order is actually the order it comes out of the printer, this is on the socket for, some reason the Flask API sorts everything. Maybe historical, since dict order was not guaranteed on Python 2 (or up to 3.6) so maybe it does it's own sorting. So, nothing I can do about that one unless I implement my own sorting on the frontend to standardise it.

There is one issue, I can no longer upload backups. I didn't touch the backup code, but maybe something it relied on got deleted 🤔.

KenLucke, 2021-05-30:
> The 'scrambled' order is actually the order it comes out of the printer, this is on the socket

I kinda figured that was probably the "native" order, although I didn't compare it to an M503

> So, nothing I can do about that one unless I implement my own sorting on the frontend to standardise it.

OK, it's merely cosmetic anyway.

> There is one issue, I can no longer upload backups. I didn't touch the backup code, but maybe something it relied on got deleted 🤔.

Interesting.  It seems to work for me, although I haven't dug into it.  I do, however, have to click the restore button twice most of the time.  I'll do some more checking when I have time.

cp2004, 2021-06-02:
RC2 out now, fixing a few issues observed in testing. New feedback ticket, #46. Hoping that it is good enough for release from that ticket.

The backup uploading has been switched from jQueryUI File Upload, to a native JS solution. The old one was a bit of a hack anyway and didn't seem to work reliably, at least in my tests. New version does seem to work every time, but of course now needs a test 😄.

closed 2021-06-02 by cp2004