cp2004/OctoPrint-EEPROM-Marlin #48: [Bug]: Capabilities re-arranging in the Firmware Info screen

andritolion, 2021-08-01:
### Description of the bug

When pressing the "Load" button in the Firmware Info tab, the Capabilities rearrange from alphabetcial order to a seemingly random order. Expected behavior is that the capabilities remain in alphabetical order.

### Plugin version

3.1.0

### OctoPrint version

1.6.1

### M503 Response

```gcode
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E93.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P4000.00 R500.00 T4000.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.30 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 W I0 J0 Z-0.19950
Recv: echo:  G29 W I1 J0 Z-0.27300
Recv: echo:  G29 W I2 J0 Z-0.43650
Recv: echo:  G29 W I0 J1 Z0.10400
Recv: echo:  G29 W I1 J1 Z0.00150
Recv: echo:  G29 W I2 J1 Z-0.19300
Recv: echo:  G29 W I0 J2 Z0.27150
Recv: echo:  G29 W I1 J2 Z0.14550
Recv: echo:  G29 W I2 J2 Z-0.08900
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H150.00 B45.00 F255
Recv: echo:  M145 S1 H190.00 B55.00 F255
Recv: echo:  M145 S2 H240.00 B70.00 F255
Recv: echo:  M145 S3 H240.00 B70.00 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P25.83 I2.15 D77.64
Recv: echo:  M304 P136.36 I26.58 D466.35
Recv: echo:; LCD Brightness:
Recv: echo:  M251 B255
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-44.00 Y-7.00 Z-1.22
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U0.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S0 D0.00
Recv: ok
```

### API Response

```json
{"backups":[{"name":"eeprom_backup-20210731-225748","time":"2021-07-31 22:57:48"}],"eeprom":{"advanced":{"command":"M205","params":{"B":20000.0,"E":5.0,"J":null,"S":0.0,"T":0.0,"X":20.0,"Y":20.0,"Z":0.3}},"autolevel":{"command":"M420","params":{"S":false,"Z":10.0}},"bed_pid":{"command":"M304","params":{"D":466.35,"I":26.58,"P":136.36}},"delta":{"command":"M665","params":{"A":null,"B":null,"C":null,"H":null,"L":null,"R":null,"S":null,"X":null,"Y":null,"Z":null}},"endstop":{"command":"M666","params":{"X":null,"Y":null,"Z":null}},"feedrate":{"command":"M203","params":{"E":25.0,"X":500.0,"Y":500.0,"Z":5.0}},"filament":{"command":"M200","params":{"D":1.75}},"filament_change":{"command":"M603","params":{"L":0.0,"U":0.0}},"filament_runout":{"command":"M412","params":{"D":0.0,"H":null,"S":false}},"home_offset":{"command":"M206","params":{"X":0.0,"Y":0.0,"Z":0.0}},"hotend_pid":{"command":"M301","params":{"D":77.64,"I":2.15,"P":25.83}},"linear":{"command":"M900","params":{"K":null}},"material1":{"command":"M145","params":{"B":45.0,"F":255.0,"H":150.0,"S":0.0}},"material2":{"command":"M145","params":{"B":55.0,"F":255.0,"H":190.0,"S":1.0}},"max_acceleration":{"command":"M201","params":{"E":1000.0,"X":500.0,"Y":500.0,"Z":100.0}},"print_acceleration":{"command":"M204","params":{"P":4000.0,"R":500.0,"T":4000.0}},"probe_offset":{"command":"M851","params":{"X":-44.0,"Y":-7.0,"Z":-1.22}},"steps":{"command":"M92","params":{"E":93.0,"X":80.0,"Y":80.0,"Z":400.0}},"tmc_current":{"command":"M906","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_hybrid":{"command":"M913","params":{"E":null,"X":null,"Y":null,"Z":null}}},"info":{"additional":{"EXTRUDER_COUNT":"1","FIRMWARE_NAME":"Marlin Marlin-JyersUI (Jul 30 2021 01:51:50)","MACHINE_TYPE":"Ender-3 V2","PROTOCOL_VERSION":"1.0","SOURCE_CODE_URL":"github.com/MarlinFirmware/Marlin","UUID":"cede2a2f-41a2-4748-9b12-c55c62f367ff"},"capabilities":{"ARCS":true,"AUTOLEVEL":true,"AUTOREPORT_POS":false,"AUTOREPORT_SD_STATUS":false,"AUTOREPORT_TEMP":true,"BABYSTEPPING":true,"BINARY_FILE_TRANSFER":false,"BUILD_PERCENT":true,"CASE_LIGHT_BRIGHTNESS":false,"CHAMBER_TEMPERATURE":false,"COOLER_TEMPERATURE":false,"EEPROM":true,"EMERGENCY_PARSER":true,"HOST_ACTION_COMMANDS":true,"LEVELING_DATA":true,"MEATPACK":true,"MOTION_MODES":false,"PRINT_JOB":true,"PROGRESS":false,"PROMPT_SUPPORT":true,"REPEAT":false,"RUNOUT":true,"SDCARD":true,"SD_WRITE":true,"SERIAL_XON_XOFF":false,"SOFTWARE_POWER":false,"THERMAL_PROTECTION":true,"TOGGLE_LIGHTS":false,"VOLUMETRIC":true,"Z_PROBE":true},"is_marlin":true,"name":"Marlin Marlin-JyersUI (Jul 30 2021 01:51:50)"}}
```

### OctoPrint logs

[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/6912298/octoprint.log)

cp2004, 2021-08-01:
I am aware that this happens, the issue is that the initial data fetched over the HTTP API is somehow sorted - I didn't sort it, and I'm not sure what does, but it is something within OctoPrint's flask/tornado server implementation. The data from the button click is reported over the websocket, which apparently doesn't sort it the same way so it remains in the same order it comes out of the printer.

I am not sure if I'm really interested in fixing the issue, as it *requires* sorting all the data as it arrives in the frontend - I would rather leave it all as it is read from the printer, but that's not an option. If someone else was to send in a PR, I would accept it but I'm not sure if I have time to work on this.

As a side note, thank you very much for completely filling in the bug report template! It is helping me to gather examples of other printer communication, as I can now throw your M503 response at my install to see what it does with it if you were reporting another kind of bug. Thanks!

cp2004, 2021-11-13:
The capabilities are now sorted alphabetically all the time - it was really not as hard as my response made it out to be, I was just lazy and have since learnt more JavaScript.

cp2004, 2021-11-14:
There's a release candidate build available if you want to test it - switch to 'Release Candidate' channel for this plugin in the software update settings in OctoPrint.

Release notes: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1
Feedback ticket: #56

cp2004, 2021-12-31:
3.2.0 has been released 🙂

closed 2021-12-31 by cp2004