cp2004/OctoPrint-EEPROM-Marlin #50: [Bug]: Editor will always fail on printers with the PASSWORD_FEATURE enabled.

R3velation, 2021-08-18:
### Description of the bug

When connecting to a printer with the password feature enabled, the editor will fail to read from the eeprom and assume that there is no printer connected at all.  This state remains regardless of when the printer is unlocked and only ends when octoprint is disconnects from the printer.  This bug will simply reoccur when the printer re-locks due to octoprint restarting it on connection.

The editor should be able to detect, respond accordingly to a locked printer, and function normally when said printer is put in an unlocked state.  Either by polling to see if the printer is unlocked, or waiting for the user to manually force a retry.

For the sake of possibly saving time (and because its not in any of the data included below), the printer is running marlin 2.0.9.1 and does have a working, enabled eeprom.

### Plugin version

3.1.0

### OctoPrint version

1.6.1

### M503 Response

```gcode
Send: N19 M503*29
Recv: echo:Printer locked! (Unlock with M511 or LCD)
Recv: ok
```

### API Response

```json
{"backups":[{"name":"eeprom_backup-20210817-172020","time":"2021-08-17 17:20:20"}],"eeprom":{"advanced":{"command":"M205","params":{"B":null,"E":null,"J":null,"S":null,"T":null,"X":null,"Y":null,"Z":null}},"autolevel":{"command":"M420","params":{"S":null,"Z":null}},"bed_pid":{"command":"M304","params":{"D":null,"I":null,"P":null}},"delta":{"command":"M665","params":{"A":null,"B":null,"C":null,"H":null,"L":null,"R":null,"S":null,"X":null,"Y":null,"Z":null}},"endstop":{"command":"M666","params":{"X":null,"Y":null,"Z":null}},"feedrate":{"command":"M203","params":{"E":null,"X":null,"Y":null,"Z":null}},"filament":{"command":"M200","params":{"D":null}},"filament_change":{"command":"M603","params":{"L":null,"U":null}},"filament_runout":{"command":"M412","params":{"D":null,"H":null,"S":null}},"home_offset":{"command":"M206","params":{"X":null,"Y":null,"Z":null}},"hotend_pid":{"command":"M301","params":{"D":null,"I":null,"P":null}},"linear":{"command":"M900","params":{"K":null}},"material1":{"command":"M145","params":{"B":null,"F":null,"H":null,"S":null}},"material2":{"command":"M145","params":{"B":null,"F":null,"H":null,"S":null}},"max_acceleration":{"command":"M201","params":{"E":null,"X":null,"Y":null,"Z":null}},"print_acceleration":{"command":"M204","params":{"P":null,"R":null,"T":null}},"probe_offset":{"command":"M851","params":{"X":null,"Y":null,"Z":null}},"steps":{"command":"M92","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_current":{"command":"M906","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_hybrid":{"command":"M913","params":{"E":null,"X":null,"Y":null,"Z":null}}},"info":{"additional":{},"capabilities":{},"is_marlin":false,"name":""}}
```

### OctoPrint logs

[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/7004419/octoprint.log)

cp2004, 2021-08-18:
Once you have unlocked the printer, it should work to hit 'refresh' in the UI to pull in the data.

I won't make the plugin poll the printer, but maybe parsing that message and showing a notice in the UI to unlock your printer first and then refresh would be sensible.

cp2004, 2021-09-02:
@R3velation - did you get a chance to test the above? Does it work fine after unlocking & refresh? I don't have this feature so I'm not able to test it.

R3velation, 2021-09-07:
Refreshing after unlocking exhibits the exact same behavior as described in the original post.

cp2004, 2021-11-13:
When the printer responds that it is locked, a warning will now be shown:

[image]

When the next M503 command is sent to the printer & goes through OK, the warning is removed - the 'refresh' (& other) buttons are not disabled.

cp2004, 2021-11-14:
There's a release candidate build available if you want to test it - switch to 'Release Candidate' channel for this plugin in the software update settings in OctoPrint.

Release notes: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1
Feedback ticket: #56

R3velation, 2021-11-18:
This only works when `PASSWORD_ON_STARTUP` is disabled.  The default when `PASSWORD_FEATURE` is enabled is that it will also have `PASSWORD_ON_STARTUP` enabled.

This shows exactly the same behavior as prior.

cp2004, 2021-11-18:
I'm going to need a full serial.log of the connection then. I wrote the detection based on what you provided in the initial issue, if that's not what always happens then I need the full thing, start to finish. Enable serial.log in the settings, connect to the printer, so whatever you need to do and then upload it here.

R3velation, 2021-11-19:
[serial.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/7566828/serial.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/7566829/octoprint.log)

And in case if you need it, an updated API response.
```json
{"backups":[],"eeprom":{"advanced":{"command":"M205","params":{"B":20000.0,"E":5.0,"J":null,"S":0.0,"T":0.0,"X":5.0,"Y":5.0,"Z":5.0}},"autolevel":{"command":"M420","params":{"S":false,"Z":null}},"bed_pid":{"command":"M304","params":{"D":null,"I":null,"P":null}},"delta":{"command":"M665","params":{"A":0.0,"B":0.0,"C":0.0,"H":335.5,"L":218.0,"R":97.4,"S":80.0,"X":0.0,"Y":0.0,"Z":0.0}},"endstop":{"command":"M666","params":{"X":0.0,"Y":0.0,"Z":0.0}},"feedrate":{"command":"M203","params":{"E":100.0,"X":100.0,"Y":100.0,"Z":100.0}},"filament":{"command":"M200","params":{"D":1.75}},"filament_change":{"command":"M603","params":{"L":650.0,"U":750.0}},"filament_runout":{"command":"M412","params":{"D":null,"H":null,"S":null}},"home_offset":{"command":"M206","params":{"X":null,"Y":null,"Z":null}},"hotend_pid":{"command":"M301","params":{"D":90.22,"I":3.55,"P":35.82}},"linear":{"command":"M900","params":{"K":null}},"material1":{"command":"M145","params":{"B":null,"F":255.0,"H":190.0,"S":0.0}},"material2":{"command":"M145","params":{"B":null,"F":255.0,"H":240.0,"S":1.0}},"max_acceleration":{"command":"M201","params":{"E":3000.0,"X":3000.0,"Y":3000.0,"Z":3000.0}},"print_acceleration":{"command":"M204","params":{"P":3000.0,"R":3000.0,"T":3000.0}},"probe_offset":{"command":"M851","params":{"X":null,"Y":null,"Z":null}},"steps":{"command":"M92","params":{"E":417.0,"X":80.0,"Y":80.0,"Z":80.0}},"tmc_current":{"command":"M906","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_hybrid":{"command":"M913","params":{"E":null,"X":null,"Y":null,"Z":null}}},"info":{"additional":{},"capabilities":{},"is_marlin":false,"name":""},"stats":{"failed":0,"filament":"","finished":0,"longest":"","prints":0,"total_time":""}}
```

cp2004, 2021-11-19:
Ok, I see what the issue is now, thank you.

The printer never responds to OctoPrint's M115 command, which should tell OP all the details about the printer, including capabilities and the firmware name. The EEPROM editor uses the data collected here by OctoPrint to activate the editor, because it can then confirm that the printer is Marlin based.

The plugin also only monitors the serial output when it is collecting data (after an M503/M501 command), to make sure it is not impacting performance at all, but M503 is only sent once we have confirmed the printer is connected & is Marlin, and so it has no knowledge of the printer's locked state. Hopefully you can see the circle created now...

I will have to work out a way of showing that the printer is connected, but locked without depending on knowing it's capabilities to activate the editor first.

cp2004, 2021-12-04:
I will get to this, eventually - struggling with time. Implemented a printer lock into the virtual printer in OctoPrint so I can fully test it & experiment, hopefully get something good made soon.

cp2004, 2021-12-24:
Published a new prerelease (3.2.0rc2) - this should work fine now, given it a good test with the actual printer lock feature to test against it is so much easier. Will go to stable as soon as possible - if there's any leftover bugs I can always publish a bugfix release.

R3velation, 2021-12-28:
Tested RC2, the odd behavior noted in RC1 is no longer present and everything is behaving exactly as it should.
I'd say this issue has been resolved.

commit 73f400a4 (2021-11-13): :sparkles: Show 'Printer locked' message

Closes #50
commit 156013cb (2021-12-24): :bug: Fix printer lock behaviour

Finally, figured out a good way to do it I think. The crucial M115 command is sent after we thought the printer.

Closes #50
closed 2021-12-28 by R3velation