cp2004/OctoPrint-EEPROM-Marlin #52: [Request]: support for editing more than one tmc driver current setting when dual, triple or quad Z drivers are configured

CBDesignS, 2021-09-09:
### What would you like implemented?

I know this may be of no use to most users but to those that have more than one independantly controlled Z axis stepper it would be of great use. (Creality Ender 5 with dual independant Z and custom built machines fall into this FR.)

Could it be added to read and adjust the current on multi Z axis stepper drivers when individual stepper drivers are used.  I Do not know if any of my ramblings comes over as it is supposed to do and I quite often confuse myself trying to write and explain things but here goes:-

I use 2 steppers with 2 tmc stepper drivers ( one on each stepper motor) on my Z axis, Marlin is configured to use 2 steppers and use G34 leveling etc, In the eeprom settings plugin it only shows one Z axis driver current so does it is only adjust the one that I can see or does it adjusts both at the same time using some hidden magic ?.

If it only adjusts say Z1 can you add an option to adjust Z2  (could be wrong with the numbering and its Z0 & Z1 as you know I aint no marlin guru) Maybe it might be possible to read the amount of marlin configured Z and  get your great eeprom plugin to match.

The only problem I see that may make it not possible is that the second driver is in what would have been a second extruder that is now reconfigured as the second Z axis stepper and I take it that is why it shows up as I1 Z550 in the M503 printout but then again the second extruder would be T1.

If this is not possible or viable please feel free to close & delete this request

Thanks in advance.
Chris

### M503 Response

```gcode
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 S0 D1.7500
echo:; Steps per unit:
echo: M92 X80.0000 Y80.0000 Z800.0000 E138.2978
echo:; Maximum feedrates (units/s):
echo:  M203 X300.0000 Y300.0000 Z10.0000 E25.0000
echo:; Maximum Acceleration (units/s2):
echo:  M201 X500.0000 Y500.0000 Z100.0000 E5000.0000
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.0000 R500.0000 T500.0000
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.0000 S0.0000 T0.0000 J0.0800
echo:; Home offset:
echo:  M206 X0.0000 Y0.0000 Z0.0000
echo:; Auto Bed Leveling:
echo:  M420 S1 Z10.0000
echo:  G29 W I0 J0 Z0.18187
echo:  G29 W I1 J0 Z0.03937
echo:  G29 W I2 J0 Z0.07625
echo:  G29 W I3 J0 Z0.06937
echo:  G29 W I4 J0 Z0.23937
echo:  G29 W I0 J1 Z0.16875
echo:  G29 W I1 J1 Z0.05687
echo:  G29 W I2 J1 Z0.03250
echo:  G29 W I3 J1 Z0.01437
echo:  G29 W I4 J1 Z0.12563
echo:  G29 W I0 J2 Z0.13875
echo:  G29 W I1 J2 Z0.07750
echo:  G29 W I2 J2 Z0.09125
echo:  G29 W I3 J2 Z0.09187
echo:  G29 W I4 J2 Z0.14625
echo:  G29 W I0 J3 Z0.14875
echo:  G29 W I1 J3 Z0.09000
echo:  G29 W I2 J3 Z0.09938
echo:  G29 W I3 J3 Z0.10938
echo:  G29 W I4 J3 Z0.20250
echo:  G29 W I0 J4 Z0.20000
echo:  G29 W I1 J4 Z0.13750
echo:  G29 W I2 J4 Z0.15188
echo:  G29 W I3 J4 Z0.20313
echo:  G29 W I4 J4 Z0.32250
echo:; Material heatup parameters:
echo:  M145 S0 H160.0000 B70.0000 F0
echo:  M145 S1 H180.0000 B70.0000 F0
echo:  M145 S2 H180.0000 B100.0000 F0
echo:  M145 S3 H180.0000 B70.0000 F0
echo:; PID settings:
echo:  M301 P25.4240 I2.5085 D64.4182
echo:  M304 P124.5676 I19.4288 D532.4437
echo:; Z-Probe Offset (mm):
echo:  M851 X-45.0000 Y-25.0000 Z-0.4900
echo:; Stepper driver current:
echo:  M906 X550 Y550 Z550
echo:  M906 I1 Z550
echo:  M906 T0 E600

echo:; Hybrid Threshold:
echo:  M913 X229 Y229 Z164
echo:  M913 I1 Z164
echo:  M913 T0 E19

echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Z
echo:  M569 S1 T0 E
```

cp2004, 2021-09-17:
First, apologies I didn't catch this issue. It seems to have slipped through as I've been really busy recently.

> In the eeprom settings plugin it only shows one Z axis driver current so does it is only adjust the one that I can see or does it adjusts both at the same time using some hidden magic ?.

I don't actually know on this one, since I lack any way of testing it & reading the Marlin code is not exactly the easiest thing to do. This plugin will send `M906 X<value> Y<value> Z<value>`, as in the line of your M503 response `M913 X229 Y229 Z164`. The second stepper current & the extruder is controlled with `M906 I1 Z<value>`, which this plugin *does not* support. I don't know if the former command will adjust both or one, but that is what is sent to the printer & displayed in the settings. There is probably even a bug here with parsing the settings out of the M503 as I think about it, but lets hope not.

Multiple Extruders/Stepper Drivers/Materials/... is not something this plugin does very well at all. The logic is very simple with how it constructs the commands, and it is not smart enough to know how to pair up extruder numbers & values.

I know that's not a very helpful answer, but hopefully it can answer the questions from the plugin's side. Thank you for providing the M503 command, as I have more and more data to reference should I try and go for implementing multiple 'items' for each command. But I can say for sure it *will not* happen soon, as I'm struggling to find time to work on OctoPrint-related code.

CBDesignS, 2021-09-17:
Thanks for replying and explaining how it`s done. on the 12864 screen when looking at the driver current settings it shows the config for X:, Y; Z; Z2; E; so certain parts can see what is I; in the 503 is then passed over as Z2; If you ever look into it and need a test machine feel free to shout, you helped me and if i can I will return the help in testing.

cp2004, 2022-12-29:
This is now available in 3.3.0rc1, which you can get by switching to 'Release Candidate' for this plugin in the software update settings.

closed 2023-01-03 by cp2004