cp2004/OctoPrint-EEPROM-Marlin #53: [Request]: Link Firmware info to config value's

spacecabbie, 2021-09-18:
### What would you like implemented?

I think it would be really nice to link the result of Firmware Info to https://marlinfw.org/meta/gcode/ to specify what a certain setting is related to.

Could help out but I am limited in programming knowledge so would need some pointers as to where to add the links.

cp2004, 2021-09-24:
This is an interesting idea - it could also be implemented with the individual EEPROM values as well. That would actually be easier than doing the firmware capabilities lines - they are very crudely just parsed straight out of the printer, it would require a list of 'known' capabilities to match against. Linking to sections of https://marlinfw.org/docs/configuration/configuration.html might also be an option.

I don't have time to do this now, but if someone were to send in a PR, I would love to review it 🙂

spacecabbie, 2021-09-25:
Wel if you would be willing to help me get started (guide me a bit on how
to write/adjust it) I am happy to get a base going for ya.

cp2004, 2021-11-13:
It was not as hard as I thought it might have been to implement this one - so I did it 🙂 

A list of the links was manually compiled from the current Marlin source, but otherwise it seems to work nicely:

[image]

cp2004, 2021-11-13:
Oh and there is also links to the command for each section: 
[image]

cp2004, 2021-11-14:
There's a release candidate build available if you want to test it - switch to 'Release Candidate' channel for this plugin in the software update settings in OctoPrint.

Release notes: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1
Feedback ticket: #56

closed 2021-12-31 by cp2004

spacecabbie, 2022-01-03:
> There's a release candidate build available if you want to test it - switch to 'Release Candidate' channel for this plugin in the software update settings in OctoPrint.
> 
> Release notes: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1 Feedback ticket: #56

This is Briliant looks awsome and works nice sorry for the late response printer was out of action. Tested it with the new BTT SKR mini e3 V3.0
Ps. some entry's are still missing how can I add those ?

deanZZZZZ, 2022-05-26:
[image]

commit 00cd04be (2021-11-13): :sparkles: Link to documentation for capabilities and editing sections

Closes #53