cp2004/OctoPrint-EEPROM-Marlin #54: [Request]: BED PID Values

skl111, 2021-11-04:
### What do you need help with?

Why does the JSON have this?
"bed_pid": {"command": "M304", "params": {"P": null, "I": null, "D": null}}
Send: M304
Recv: echo:  M304 P180.15 I35.19 D614.93
Is what is reported via terminal?

### M503 Response

```gcode
Send: M503
Recv:  T:19.22 /0.00 B:23.55 /0.00 @:0 B@:0
Recv: echo:; Linear Units:
Recv:   G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z800.00 E142.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z10.00 ; Leveling ON
Recv: echo:  G29 W I0 J0 Z-0.07876
Recv: echo:  G29 W I1 J0 Z0.09749
Recv: echo:  G29 W I2 J0 Z0.09149
Recv: echo:  G29 W I3 J0 Z0.20499
Recv: echo:  G29 W I4 J0 Z0.14749
Recv: echo:  G29 W I0 J1 Z-0.11601
Recv: echo:  G29 W I1 J1 Z0.00024
Recv: echo:  G29 W I2 J1 Z0.00874
Recv: echo:  G29 W I3 J1 Z0.10774
Recv: echo:  G29 W I4 J1 Z0.10124
Recv: echo:  G29 W I0 J2 Z-0.14101
Recv: echo:  G29 W I1 J2 Z-0.01276
Recv: echo:  G29 W I2 J2 Z0.01599
Recv: echo:  G29 W I3 J2 Z0.07374
Recv: echo:  G29 W I4 J2 Z0.06374
Recv: echo:  G29 W I0 J3 Z-0.23376
Recv: echo:  G29 W I1 J3 Z-0.08251
Recv: echo:  G29 W I2 J3 Z-0.05126
Recv: echo:  G29 W I3 J3 Z0.06249
Recv: echo:  G29 W I4 J3 Z0.06124
Recv: echo:  G29 W I0 J4 Z-0.32926
Recv: echo:  G29 W I1 J4 Z-0.15401
Recv: echo:  G29 W I2 J4 Z-0.15751
Recv: echo:  G29 W I3 J4 Z0.03024
Recv: echo:  G29 W I4 J4 Z0.07524
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H220.00 B60.00 F255
Recv: echo:  M145 S1 H235.00 B75.00 F255
Recv: echo:; Hotend PID:
Recv: echo:  M301 P21.73 I1.54 D76.55
Recv: echo:; Bed PID:
Recv: echo:echo:  M304 P180.15 I35.19 D614.93
Recv: echo:; Controller Fan:
Recv: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-48.00 Y-10.00 Z-3.99 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L200.00 U120.00 ; (mm)
Recv: ok
```

cp2004, 2021-11-04:
> Recv: echo:; Bed PID:
> Recv: echo:echo:  M304 P180.15 I35.19 D614.93

The double echo here is why. None of the rest of the lines have it. It's probably a bug in the firmware you're using and should be fixed there to remove the doubling.

skl111, 2021-11-04:
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Oct 21 2021 23:36:20) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin

Looks like I will be submitting a new bug,  ugh!

cp2004, 2021-11-06:
The mentioned PR in Marlin should fix this.

https://github.com/MarlinFirmware/Marlin/pull/23084

closed 2021-11-06 by cp2004

skl111, 2021-11-07:
I tried the change, via a patch and it works on the E5+ and it answered another question I had which was why the BED PID values were not showing up in the menu to change! Thanks!