cp2004/OctoPrint-EEPROM-Marlin #56: [Feedback] Marlin EEPROM Editor 3.2.0rc1 Feedback

cp2004, 2021-11-14:
It's here, 3.2.0rc1 Release Candidate!

Be sure to check out the [release candidate release notes](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/releases/tag/3.2.0rc1) here for details of all the changes.

Please test this release, and let me know what you think. If all goes well this will be released at some point next week, so feedback is important to make sure that everything is good! An 'All is working well' is great feedback, since it lets me know people are actually testing this release, even if there are no issues to report.

If you find a bug please use the bug report template and send across the logs, so that I can get it fixed as quickly as possible.

cp2004, 2021-12-31:
3.2.0 has been released 🙂

closed 2021-12-31 by cp2004

adamsyed10, 2023-09-14:
Hello, I would first like to thank you for the amazing work that you've done. I have a decent amount of monoprice mini delta v2s that I'm donating to a local school's computer lab. However, I am not able to figure out how to reset the "Printed Time" menu to zero. Some of the printers have glitched out printed times and I don't want to look like they're getting printers that have been used 10,000 hours. I want to be able to easily reset "Printed Time" when I factory reset them all. The printers are all in good working condition and have been tested.  Please let me know if you are able to figure out a way. I am willing to send you one of the delta mini v2s as compensation or can pay cash for your time. You can email me at adamsyed10@gmail.com or message me a good way to contact you.  Thanks for reading this and have a great day!  

Best Regards,
Adam