cp2004/OctoPrint-EEPROM-Marlin #58: CR-10 Smart printer isn't recognized as Marlin firmware

kevincrombie, 2021-12-06:
This is the error: Firmware reports name as: 1.0.10

This plugin only works with Marlin Firmware, sorry!

cp2004, 2021-12-06:
You need to fix your firmware name - it's not reporting itself as Marlin so the plugin is (correctly) not enabling the editor. The firmware name must have 'Marlin' in it somewhere.

cp2004, 2021-12-16:
Since this is not an issue with the plugin, it will be closed.

closed 2021-12-16 by cp2004

kevincrombie, 2021-12-16:
Makes sense, If I reflash the firmware how would I rename the software name in the .hex file it’s flashed with?

It’s an awesome plug-in by the way! I use it with the cr-10s pro v2 and it’s been very helpful. The Cr-10 smart is the printer it won’t work with because the name is just the numerical version of the software (1.0.10). Whereas the other has name of printer_Marlin_version.

Thanks for your help,

Kevin

cp2004, 2021-12-16:
The firmware name is hardcoded in Marlin, at least in current versions ([see here](https://github.com/MarlinFirmware/Marlin/blob/170f77fada009bcd77b02edf7b5d55d5173b00e9/Marlin/src/gcode/host/M115.cpp#L52)) - you can't edit hex files, it would have to be recompiled. If you downloaded the firmware from the manufacturer, then they should provide the source code files so you could compile it yourself. But knowing Creality it is not uncommon for them to violate the GPL license...

kevincrombie, 2021-12-16:
Creality doesn’t seem to follow both internal and external policies or laws. I think I’ve finally learned my lesson to never buy their products again. Hopefully they’ll provide me the source code now I’ve requested it. Again, thank you so much for your help and apologies for bugging you with my ignorance.

Kevin