cp2004/OctoPrint-EEPROM-Marlin #61: [Request]: Automatic scheduled EEPROM backups

rupertbenbrook, 2022-04-14:
### What would you like implemented?

I find myself forgetting to backup my EEPROM configuration before updating the Marlin firmware, then finding this requires an EEPROM reset, and I loose some settings I've changed in the meantime because I have to go back to an older backup. It would be great if I could set a regular automatic backup of the EEPROM config, so I don't have to remember. For example, a daily backup that executes once a day, if OctoPrint is running. Optionally if the backup hasn't changed then the previous auto backup could be deleted, or a maximum history of old backup kept, so as not to overfill the backups.

### M503 Response

```gcode
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E430.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R2000.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.04
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: echo:  G29 W I0 J0 Z0.11600
Recv: echo:  G29 W I1 J0 Z0.37800
Recv: echo:  G29 W I2 J0 Z0.25550
Recv: echo:  G29 W I3 J0 Z0.45200
Recv: echo:  G29 W I4 J0 Z0.35850
Recv: echo:  G29 W I5 J0 Z0.40350
Recv: echo:  G29 W I6 J0 Z0.37000
Recv: echo:  G29 W I0 J1 Z-0.12600
Recv: echo:  G29 W I1 J1 Z0.07700
Recv: echo:  G29 W I2 J1 Z-0.03700
Recv: echo:  G29 W I3 J1 Z0.18000
Recv: echo:  G29 W I4 J1 Z0.09650
Recv: echo:  G29 W I5 J1 Z0.15000
Recv: echo:  G29 W I6 J1 Z0.12900
Recv: echo:  G29 W I0 J2 Z-0.29200
Recv: echo:  G29 W I1 J2 Z-0.06400
Recv: echo:  G29 W I2 J2 Z-0.17850
Recv: echo:  G29 W I3 J2 Z0.03800
Recv: echo:  G29 W I4 J2 Z-0.04000
Recv: echo:  G29 W I5 J2 Z0.01250
Recv: echo:  G29 W I6 J2 Z-0.00100
Recv: echo:  G29 W I0 J3 Z-0.28700
Recv: echo:  G29 W I1 J3 Z-0.07600
Recv: echo:  G29 W I2 J3 Z-0.16900
Recv: echo:  G29 W I3 J3 Z0.05400
Recv: echo:  G29 W I4 J3 Z-0.00550
Recv: echo:  G29 W I5 J3 Z0.06750
Recv: echo:  G29 W I6 J3 Z0.05700
Recv: echo:  G29 W I0 J4 Z-0.25750
Recv: echo:  G29 W I1 J4 Z-0.03700
Recv: echo:  G29 W I2 J4 Z-0.15100
Recv: echo:  G29 W I3 J4 Z0.06200
Recv: echo:  G29 W I4 J4 Z-0.01300
Recv: echo:  G29 W I5 J4 Z0.04250
Recv: echo:  G29 W I6 J4 Z0.04200
Recv: echo:  G29 W I0 J5 Z-0.24200
Recv: echo:  G29 W I1 J5 Z-0.00850
Recv: echo:  G29 W I2 J5 Z-0.09050
Recv: echo:  G29 W I3 J5 Z0.15750
Recv: echo:  G29 W I4 J5 Z0.11350
Recv: echo:  G29 W I5 J5 Z0.21100
Recv: echo:  G29 W I6 J5 Z0.24600
Recv: echo:  G29 W I0 J6 Z-0.33450
Recv: echo:  G29 W I1 J6 Z-0.04200
Recv: echo:  G29 W I2 J6 Z-0.10650
Recv: echo:  G29 W I3 J6 Z0.17450
Recv: echo:  G29 W I4 J6 Z0.15350
Recv: echo:  G29 W I5 J6 Z0.28100
Recv: echo:  G29 W I6 J6 Z0.32400
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B65.00 F0
Recv: echo:  M145 S1 H240.00 B90.00 F0
Recv: echo:; Hotend PID:
Recv: echo:  M301 P33.33 I1.93 D144.03
Recv: echo:; Bed PID:
Recv: echo:  M304 P26.34 I4.66 D99.27
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-37.00 Y0.00 Z-1.85 ; (mm)
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U450.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok
```

cp2004, 2022-05-10:
Just to acknowledge - I have seen this, currently undecided if it is something I will include but I can see the usefulness of having it. 

I think proper custom scheduling can get complicated to code, so I wondered if maybe automatically saving a backup when a change is made would be useful? (it would possibly be simpler to create!). Obviously with a limit to the last X number of changes. Would that be useful to you? Or do you not use the editor that often and just use it for backups?

As I said in the other issue here - I have not got any time to look into this for the coming 2 months, but after that I may be able to look at it.

rupertbenbrook, 2022-07-24:
@cp2004 sorry for taking so long to respond. Yes, I think a simple backup whenever a change was made would be a step in the right direction. As you say, you probably want to have a backup each time, but perhaps a maximum number of backups.