cp2004/OctoPrint-EEPROM-Marlin #62: Retrieve and Set Stealthchop for TMC Drivers[Request]:

skl111, 2022-05-02:
### What would you like implemented?

It would be great if M569 was integrated so you can see the Stealthchop mode Status and back it up before you flash firmware so you could restore that config.

### M503 Response

```gcode
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop
Recv: E driver mode:	spreadCycle
Recv: ok
```

cp2004, 2022-05-10:
Does this data show up in an M503 command? Currently, pretty much everything that is included with this plugin is parsed out of that command and I would like to avoid a sending a whole list of commands if I can.

If you can post the full M503 response that would be great so I can have a copy to test against, but I will also say I have not got any time to add this for the coming 2 months, but after that I may be able to look at it.

skl111, 2022-05-11:
Yes it does. m569 near the bottom

`Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z800.00 E142.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: echo:  G29 W I0 J0 Z0.06125
Recv: echo:  G29 W I1 J0 Z0.10650
Recv: echo:  G29 W I2 J0 Z0.11625
Recv: echo:  G29 W I3 J0 Z0.10900
Recv: echo:  G29 W I4 J0 Z0.10475
Recv: echo:  G29 W I5 J0 Z0.08525
Recv: echo:  G29 W I6 J0 Z0.08900
Recv: echo:  G29 W I7 J0 Z0.09250
Recv: echo:  G29 W I8 J0 Z0.04875
Recv: echo:  G29 W I0 J1 Z0.03400
Recv: echo:  G29 W I1 J1 Z0.07400
Recv: echo:  G29 W I2 J1 Z0.06600
Recv: echo:  G29 W I3 J1 Z0.08775
Recv: echo:  G29 W I4 J1 Z0.09000
Recv: echo:  G29 W I5 J1 Z0.08750
Recv: echo:  G29 W I6 J1 Z0.07875
Recv: echo:  G29 W I7 J1 Z0.06400
Recv: echo:  G29 W I8 J1 Z0.03725
Recv: echo:  G29 W I0 J2 Z0.01625
Recv: echo:  G29 W I1 J2 Z0.05625
Recv: echo:  G29 W I2 J2 Z0.05750
Recv: echo:  G29 W I3 J2 Z0.05375
Recv: echo:  G29 W I4 J2 Z0.07375
Recv: echo:  G29 W I5 J2 Z0.06850
Recv: echo:  G29 W I6 J2 Z0.06625
Recv: echo:  G29 W I7 J2 Z0.05875
Recv: echo:  G29 W I8 J2 Z0.00200
Recv: echo:  G29 W I0 J3 Z0.01125
Recv: echo:  G29 W I1 J3 Z0.06775
Recv: echo:  G29 W I2 J3 Z0.06000
Recv: echo:  G29 W I3 J3 Z0.06125
Recv: echo:  G29 W I4 J3 Z0.08150
Recv: echo:  G29 W I5 J3 Z0.04125
Recv: echo:  G29 W I6 J3 Z0.05475
Recv: echo:  G29 W I7 J3 Z0.02125
Recv: echo:  G29 W I8 J3 Z-0.00600
Recv: echo:  G29 W I0 J4 Z0.01000
Recv: echo:  G29 W I1 J4 Z0.05250
Recv: echo:  G29 W I2 J4 Z0.08025
Recv: echo:  G29 W I3 J4 Z0.08225
Recv: echo:  G29 W I4 J4 Z0.06500
Recv: echo:  G29 W I5 J4 Z0.04475
Recv: echo:  G29 W I6 J4 Z0.03000
Recv: echo:  G29 W I7 J4 Z0.01775
Recv: echo:  G29 W I8 J4 Z-0.02625
Recv: echo:  G29 W I0 J5 Z0.04650
Recv: echo:  G29 W I1 J5 Z0.05000
Recv: echo:  G29 W I2 J5 Z0.06975
Recv: echo:  G29 W I3 J5 Z0.09150
Recv: echo:  G29 W I4 J5 Z0.07000
Recv: echo:  G29 W I5 J5 Z0.02500
Recv: echo:  G29 W I6 J5 Z0.06375
Recv: echo:  G29 W I7 J5 Z0.00875
Recv: echo:  G29 W I8 J5 Z-0.03100
Recv: echo:  G29 W I0 J6 Z0.02225
Recv: echo:  G29 W I1 J6 Z0.06600
Recv: echo:  G29 W I2 J6 Z0.09475
Recv: echo:  G29 W I3 J6 Z0.08125
Recv: echo:  G29 W I4 J6 Z0.05875
Recv: echo:  G29 W I5 J6 Z0.04375
Recv: echo:  G29 W I6 J6 Z0.04875
Recv: echo:  G29 W I7 J6 Z0.01250
Recv: echo:  G29 W I8 J6 Z-0.03000
Recv: echo:  G29 W I0 J7 Z0.03275
Recv: echo:  G29 W I1 J7 Z0.08150
Recv: echo:  G29 W I2 J7 Z0.11150
Recv: echo:  G29 W I3 J7 Z0.07100
Recv: echo:  G29 W I4 J7 Z0.08750
Recv: echo:  G29 W I5 J7 Z0.04850
Recv: echo:  G29 W I6 J7 Z0.07025
Recv: echo:  G29 W I7 J7 Z0.03625
Recv: echo:  G29 W I8 J7 Z-0.02000
Recv: echo:  G29 W I0 J8 Z0.06150
Recv: echo:  G29 W I1 J8 Z0.12125
Recv: echo:  G29 W I2 J8 Z0.13350
Recv: echo:  G29 W I3 J8 Z0.11625
Recv: echo:  G29 W I4 J8 Z0.10225
Recv: echo:  G29 W I5 J8 Z0.07900
Recv: echo:  G29 W I6 J8 Z0.08275
Recv: echo:  G29 W I7 J8 Z0.09150
Recv: echo:  G29 W I8 J8 Z0.00375
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H220.00 B60.00 F255
Recv: echo:  M145 S1 H235.00 B75.00 F255
Recv: echo:; Hotend PID:
Recv: echo:  M301 P16.96 I1.27 D56.65
Recv: echo:; Bed PID:
Recv: echo:  M304 P121.71 I23.05 D428.41
Recv: echo:; Controller Fan:
Recv: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-40.00 Y-10.00 Z-2.22 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.20
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L200.00 U100.00 ; (mm)
Recv: ok`

`Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop
Recv: E driver mode:	spreadCycle`

cp2004, 2023-02-26:
Just to follow up on why this has not been added - the driver modes are defined a bit differently to almost every other EEPROM command.

S1: Set StealthChop mode
S0: Set SpreadCycle mode

This can be set for the different axes individually or multiple in one go, which is more difficult to fit in to the common data structure/definitions every other command follows. Additionally, in the M503 response it doesn't actually include the E configuration (perhaps because it is spreadcycle and the others are stealthchop) which makes it hard for this plugin to work out whether to show E values or not (Also applies to a printer that might have multiple tools, multiple Z axis steppers, etc.) 

If I find time (there are other plugins higher up the TODO list), I may have a look to see if there is a way around this but I can't say if I will be able to add this.

skl111, 2023-02-26:
Yep, your correct in that when its SpreadCycle it doesn't show.
They correspond that with the S0 having it set to off and the S1 having it set to on for the other motors.

I understand there are other updates above this task, and I can continue to work around not having this feature.