cp2004/OctoPrint-EEPROM-Marlin #63: [Bug]: PluginManager typo -> link to gihtub

erstert3st, 2022-05-10:
### Description of the bug

Go to pluginmanager -> search your plugin and click on Hompage
now -> https://github.com/cp2004/OctoPrint-EEPROM-Marlin.
should be: https://github.com/cp2004/OctoPrint-EEPROM-Marlin

### Plugin version

newes

### OctoPrint version

newes

### M503 Response

```gcode
Typo
```

### API Response

```json
Typo
```

### OctoPrint logs

Typo

cp2004, 2022-05-10:
> Plugin version
> newes

Please actually specify the version, rather than 'newest'. This was fixed in 3.2.0 already.

erstert3st, 2022-05-10:
Plugin search (3.2.0) -> Link working!
Plugin installed manually or via URL ( 3.2.0)=-> Link in installed plugin list working!

**Plugin installed via plugin-manager (3.2.0)  
Installed plugin list  (plugin-manager)-> Link  not working**

cp2004, 2022-05-10:
I have no idea why that is since the download URLs are exactly the same for both methods of installing it. As far as I can tell it is fixed in the latest version that I have on GitHub.

closed 2022-05-10 by cp2004