cp2004/OctoPrint-EEPROM-Marlin #64: [Request]: Only FIRMWARE INFO in the left column

rashn, 2022-05-16:
### What do you need help with?

What wrong? Only FIRMWARE INFO in the left column.

[image]

cp2004, 2022-05-16:
And what happens when you press 'Load'?

rashn, 2022-05-16:
Reloaded main frame
[https://youtube.com/shorts/Mqtg_7G5_uw?feature=share](url)

cp2004, 2022-05-16:
In that case, please provide the M503 response (as in the issue form) and also the octoprint.log file.

rashn, 2022-05-16:
Send: M503
Recv:   G21    ; Units in mm (mm)
Recv:   M149 C ; Units in Celsius
Recv: 
Recv:   M92 X200.00 Y200.00 Z800.00 E190.00
Recv:   M203 X400.00 Y400.00 Z3.00 E25.00
Recv:   M201 X500.00 Y500.00 Z10.00 E7000.00
Recv:   M204 P600.00 R3000.00 T3000.00
Recv:   M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv:   M206 X0.00 Y0.00 Z0.00
Recv:   M420 S1 Z10.00
Recv:   M145 S0 H220 B60 F0
Recv:   M145 S1 H240 B110 F0
Recv:   M301 P22.90 I1.48 D88.50
Recv:   M304 P270.13 I43.25 D421.81
Recv:   M851 Z-1.90
Recv:   M906 X1000 Y1000 Z1050
Recv:   M906 T0 E1000
Recv: 
Recv:   M913 X100 Y100 Z10
Recv:   M913 T0 E30
Recv: 
Recv:   M569 S1 X Y Z
Recv:   M569 S1 T0 E
Recv: ok

rashn, 2022-05-16:
[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/8702297/octoprint.log)

rashn, 2022-05-17:
One more things
[image]
:

rashn, 2022-05-24:
Hello dear friend! Any news about my problem solving?

cp2004, 2022-06-28:
I am sorry for the long delay here. I have been busy and forgot about this issue.

The M503 response you have provided tells the story - I have looked again at the required format. There is a missing `echo: ` at the start of the line which is present in mainline Marlin firmware (not forks). I am hesitant to add support for other forks because it will get messy if different people keep changing the format slightly.

Although the firmware info section says 'Source code URL https://github.com/MarlinFirmware/Marlin' I cannot find any evidence of this behaviour in the code there, so this must be a fork of the firmware. Where did you get the firmware on your printer from?

rashn, 2022-07-14:
This is clean Marlin-bugfix-2.0.0 version of firmware https://github.com/MarlinFirmware/Marlin

cp2004, 2022-07-15:
This is the response I get from the current bugfix-2.1.0 version:
```
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E500.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.01
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H180.00 F0
Recv: echo:  M145 S1 H240.00 F0
Recv: echo:; Hotend PID:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:; Retract (S<length> F<feedrate> Z<lift>):
Recv: echo:  M207 S3.00 W13.00 F2700.00 Z0.00
Recv: echo:; Recover (S<length> F<feedrate>):
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:; Auto-Retract (S<enable>):
Recv: echo:  M209 S0
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: ok
```
It is different from yours, with the `echo` and extra comments. I've never seen a Marlin build that, and when looking at the code I can't see a way that the `echo` and all of the comments are missed out.

Can you share the source of the build somehow? Did you compile it yourself? Maybe just the `Configuration.h` and `Configuration_adv.h` files in a zip will work.

cp2004, 2022-08-21:
@rashn do you still need help with this?

rashn, 2022-08-22:
[marlin.zip](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/9394645/marlin.zip)
Yes! Still need help. Thank you!

rubienr, 2023-08-28:
I observe the same issue with the `bugfix-2.1.x` branch + EEPRPOM Editor Version 3.3.0.  Here an example for `echo` with extra comment: in `M412.cpp` in method  [void GcodeSuite::M412_report(const bool forReplay=true)](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/gcode/feature/runout/M412.cpp#L69) which results in `Recv: echo:; Filament runout sensor:`.

The whole output from OctoPrint terminal: 
```
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0
Recv: echo:  M200 T0 D1.75
Recv: echo:  M200 T1 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z1600.00 E282.16
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z8.00 E150.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z15.00 E35000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P600.00 R1000.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.01
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Hotend offsets:
Recv: echo:echo:  M218 T1 X-41.00 Y0.00 Z0.000
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S1 Z10.00 ; Leveling ON
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: echo:Active Mesh Slot 0
Recv: echo:EEPROM can hold 15 meshes.
Recv: 
Recv: echo:; Servo Angles:
Recv: echo:echo:  M281 P1 L140 U55
Recv: echo:; Z Auto-Align:
Recv: echo:  M422 S1 X8.00 Y185.00
Recv: echo:  M422 S2 X369.00 Y185.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H180.00 B70.00 F0
Recv: echo:  M145 S1 H240.00 B110.00 F0
Recv: echo:; Bed PID:
Recv: echo:  M304 P200.62 I34.83 D770.37
Recv: echo:; Controller Fan:
Recv: echo:  M710 S191 I0 A1 D30 ; (74% 0%)
Recv: echo:; Retract (S<length> F<feedrate> Z<lift>):
Recv: echo:  M207 S1.00 W20.00 F1200.00 Z0.00
Recv: echo:; Recover (S<length> F<feedrate>):
Recv: echo:  M208 S0.00 W0.13 F9000.00
Recv: echo:; Auto-Retract (S<enable>):
Recv: echo:  M209 S0
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-20.50 Y55.50 Z-2.50 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X1500 Y2000 Z800
Recv: echo:  M906 I1 Z800
Recv: echo:  M906 T0 E1200
Recv: echo:  M906 T1 E1200
Recv: echo:; Hybrid Threshold:
Recv: echo:  M913 X2470 Y1098 Z494
Recv: echo:  M913 I2 Z494
Recv: echo:  M913 T0 E90
Recv: echo:  M913 T1 E90
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 I1 Z
Recv: echo:  M569 S1 T0 E
Recv: echo:  M569 S1 T1 E
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.10
Recv: echo:; Filament load/unload:
Recv: echo:  M603 T0 L2.00 U0.00 ; (mm)
Recv: echo:  M603 T1 L2.00 U0.00 ; (mm)
Recv: echo:; Tool-changing:
Recv: echo:  M217 Z2.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: echo:; Model predictive control:
Recv: echo:  M306 E0 P40.00 C16.43 R0.1404 A0.0752 F0.1210 H0.0056
Recv: echo:  M306 E1 P40.00 C16.51 R0.2082 A0.0805 F0.0823 H0.0056
Recv: ok P15 B7
```