cp2004/OctoPrint-EEPROM-Marlin #65: [Bug]: Doesn't work with Snapmaker 2.0 A350

DavidKuhn136, 2022-06-24:
### Description of the bug

When I tried to load the eeprom for the snapmaker 2.0 a350, it didn't pull in any information.  I can only see "Firmware Info"

If i run M503, it pulls back all the information in the terminal, but I can't get the EEPROM editor to work.  I have an 11x11 grid for bed calibration set on the snapmaker, so not sure if that is the issue.

As an FYI, I also have this installed on another instance with an Ender 3 and it is amazing!

this is what is coming back from "Firmware info":
Firmware Name: Marlin SM2-4.4.16 (Github)
Source Code Url: https://github.com/whimsycwd/SnapmakerMarlin
Protocol Version: 1.0

### Plugin version

3.2.0

### OctoPrint version

1.8.1

### M503 Response

```gcode
Send: M503
Recv:   G21 ; (mm)
Recv: 
Recv:   M200 D3.00
Recv:   M200 D0
Recv:   M92 X400.00 Y400.00 Z400.00 B888.89 E258.79
Recv:   M203 X120.00 Y120.00 Z40.00 E25.00
Recv:   M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv:   M204 P750.00 R1000.00 T1000.00
Recv:   M205 B20000.00 S0.00 T0.00 P0.05 L3.00 C0.05 J0.02
Recv:   M206 X-19.00 Y-10.00 Z0.00
Recv:   M420 S1 Z0.00
Recv:   G29 W I0 J0 Z5.94250
Recv:   G29 W I1 J0 Z6.18625
Recv:   G29 W I2 J0 Z6.30875
Recv:   G29 W I3 J0 Z6.15250
Recv:   G29 W I4 J0 Z5.75625
Recv:   G29 W I0 J1 Z6.68125
Recv:   G29 W I1 J1 Z6.82875
Recv:   G29 W I2 J1 Z6.86250
Recv:   G29 W I3 J1 Z6.74750
Recv:   G29 W I4 J1 Z6.31875
Recv:   G29 W I0 J2 Z6.92250
Recv:   G29 W I1 J2 Z7.04750
Recv:   G29 W I2 J2 Z7.00000
Recv:   G29 W I3 J2 Z6.84875
Recv:   G29 W I4 J2 Z6.48875
Recv:   G29 W I0 J3 Z6.76375
Recv:   G29 W I1 J3 Z6.89625
Recv:   G29 W I2 J3 Z6.85000
Recv:   G29 W I3 J3 Z6.74000
Recv:   G29 W I4 J3 Z6.42500
Recv:   G29 W I0 J4 Z6.22625
Recv:   G29 W I1 J4 Z6.26000
Recv:   G29 W I2 J4 Z6.32250
Recv:   G29 W I3 J4 Z6.26125
Recv:   G29 W I4 J4 Z6.06500
Recv:   M301 P13.00 I0.10 D17.00
Recv:   M851 Z1.00
Recv:   M900 K0.07
Recv: ok
```

### API Response

```json
{"backups":[{"name":"eeprom_backup-20220624-185421","time":"2022-06-24 18:54:21"}],"eeprom":{"advanced":{"command":"M205","params":{"B":null,"E":null,"J":null,"S":null,"T":null,"X":null,"Y":null,"Z":null}},"autolevel":{"command":"M420","params":{"S":null,"Z":null}},"bed_pid":{"command":"M304","params":{"D":null,"I":null,"P":null}},"delta":{"command":"M665","params":{"A":null,"B":null,"C":null,"H":null,"L":null,"R":null,"S":null,"X":null,"Y":null,"Z":null}},"endstop":{"command":"M666","params":{"X":null,"Y":null,"Z":null}},"feedrate":{"command":"M203","params":{"E":null,"X":null,"Y":null,"Z":null}},"filament":{"command":"M200","params":{"D":null}},"filament_change":{"command":"M603","params":{"L":null,"U":null}},"filament_runout":{"command":"M412","params":{"D":null,"H":null,"S":null}},"home_offset":{"command":"M206","params":{"X":null,"Y":null,"Z":null}},"hotend_pid":{"command":"M301","params":{"D":null,"I":null,"P":null}},"linear":{"command":"M900","params":{"K":null}},"material1":{"command":"M145","params":{"B":null,"F":null,"H":null,"S":null}},"material2":{"command":"M145","params":{"B":null,"F":null,"H":null,"S":null}},"max_acceleration":{"command":"M201","params":{"E":null,"X":null,"Y":null,"Z":null}},"print_acceleration":{"command":"M204","params":{"P":null,"R":null,"T":null}},"probe_offset":{"command":"M851","params":{"X":null,"Y":null,"Z":null}},"steps":{"command":"M92","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_current":{"command":"M906","params":{"E":null,"X":null,"Y":null,"Z":null}},"tmc_hybrid":{"command":"M913","params":{"E":null,"X":null,"Y":null,"Z":null}}},"info":{"additional":{"EXTRUDER_COUNT":"1","FIRMWARE_NAME":"Marlin SM2-4.4.16 (Github)","MACHINE_TYPE":"GD32F305VGT6","PROTOCOL_VERSION":"1.0","SOURCE_CODE_URL":"https://github.com/whimsycwd/SnapmakerMarlin","UUID":"cede2a2f-41a2-4748-9b12-c55c62f367ff"},"capabilities":{"AUTOLEVEL":false,"AUTOREPORT_SD_STATUS":false,"AUTOREPORT_TEMP":true,"BINARY_FILE_TRANSFER":false,"BUILD_PERCENT":false,"CASE_LIGHT_BRIGHTNESS":false,"CHAMBER_TEMPERATURE":false,"EEPROM":true,"EMERGENCY_PARSER":false,"LEVELING_DATA":true,"MOTION_MODES":false,"PRINT_JOB":true,"PROGRESS":false,"PROMPT_SUPPORT":false,"SERIAL_XON_XOFF":false,"SOFTWARE_POWER":false,"THERMAL_PROTECTION":true,"TOGGLE_LIGHTS":false,"VOLUMETRIC":true,"Z_PROBE":true},"is_marlin":true,"name":"Marlin SM2-4.4.16 (Github)"}}
```

### OctoPrint logs

[octoprint.log](https://github.com/cp2004/OctoPrint-EEPROM-Marlin/files/8980686/octoprint.log)

cp2004, 2022-06-28:
The problem here is that the firmware has messed with the format of the response.

A typical M503 response looks like this (without the `Recv: ` in there):
```
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.7500
echo:; Steps per unit:
echo: M92 X200.0000 Y200.0000 Z400.0000 E415.0000
echo:; Maximum feedrates (units/s):
echo:  M203 X400.0000 Y400.0000 Z8.0000 E50.0000
echo:; Maximum Acceleration (units/s2):
echo:  M201 X2000.0000 Y2000.0000 Z100.0000 E10000.0000
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P800.0000 R10000.0000 T2000.0000
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.0000 S0.0000 T0.0000 X8.0000 Y8.0000 Z0.3000 E5.0000
echo:; Home offset:
echo:  M206 X0.0000 Y0.0000 Z0.0000
echo:; Unified Bed Leveling:
echo:  M420 S1 Z10.0000
Unified Bed Leveling System v1.01 active
echo:; Active Mesh Slot: 0
echo:; EEPROM can hold 3 meshes.
echo:; Material heatup parameters:
echo:  M145 S0 H190.0000 B60.0000 F0
echo:  M145 S1 H210.0000 B70.0000 F0
echo:; PID settings:
echo:  M301 P19.0986 I1.3998 D65.1453
echo:  M304 P46.3000 I9.0700 D157.5700
echo:; Z-Probe Offset (mm):
echo:  M851 X-29.0000 Y1.0000 Z-1.9250
echo:; Stepper driver current:
echo:  M906 X1200 Y1200 Z400
echo:  M906 I1 Z400
echo:  M906 T0 E900
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Z
echo:; Linear Advance:
echo:  M900 K0.0400
echo:; Filament load/unload lengths:
echo:  M603 L25.0000 U75.0000
echo:; Filament runout sensor:
echo:  M412 S0
ok P63 B31
```

The key difference is the missing `echo:` on your response. This is present in the mainline Marlin firmware, and I am hesitant to add support for other forks because it will get messy. You should probably ask the maintainers of the firmware you have mentioned not to mess with the gcode's output.

cp2004, 2022-07-07:
@DavidKuhn136 did the above comment clear things up for you?

cp2004, 2022-07-14:
I am considering this issue solved - if you have more questions please ask.

closed 2022-07-14 by cp2004