cp2004/OctoPrint-EEPROM-Marlin #66: [Request]: When enabled in Firmware, Add M306 data to Temperature tab

ImagineerNL, 2022-06-27:
### What would you like implemented?

See last line in the M503 response below.

More info on Model Predictive Temperature Control: https://marlinfw.org/docs/gcode/M306.html

Also, probably since in my previous firmware, PID was enabled for the hotend (must be disabled for the MPCTEMP to work), the EEPROM Editor still shows the Hotend PID data even though it isnt in the M503.

[image]

Both for PID tuning and for MPC tuning, would be a great feature to show the (link to) the gcode needed to tune those parameters, as these arent values you tune by hand.

### M503 Response

```gcode
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E133.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P800.00 R1000.00 T800.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X-3.00 Y-10.00 Z0.00
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S1 Z10.00 ; Leveling ON
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: echo:Active Mesh Slot -1
Recv: echo:EEPROM can hold 7 meshes.
Recv: 
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B110.00 F255
Recv: echo:; Bed PID:
Recv: echo:  M304 P50.71 I9.88 D173.43
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-42.00 Y-14.00 Z-2.95 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z800
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L350.00 U400.00 ; (mm)
Recv: echo:; Model predictive control:
Recv: echo:  M306 E0 P40.00 C13.52 R0.4295 A0.0835 F0.0978 H0.0056
Recv: ok P15 B3
```

cp2004, 2022-06-28:
I've added support for this to the data model. I am not sure when it will be released, it depends if I start tackling the other issues which could be more work.

> Both for PID tuning and for MPC tuning, would be a great feature to show the (link to) the gcode needed to tune those parameters, as these arent values you tune by hand.

There should be a small blue (i) button beside the entries with links to the configuration gcode. For MPC there is one gcode for everything, but for PID it is slightly different with separate commands.

ImagineerNL, 2022-06-28:
nope, no i button
[image]

I do know where to find the gcode to do a PID, but every time i have to look it up ;)

cp2004, 2022-06-28:
Hmm, this is what I have on my end:

[image]

Can you confirm which version of the plugin you have installed? It's either in the plugin manager, software update or at the bottom of the plugin's settings page.

ImagineerNL, 2022-06-29:
Version 3.2.0
newest according to both octoprint and github

my setup: Ender 3 with SKR Mini e3 V2 on Marlin 2.1.xx bugfix, connected through octoprint.
Have UI customizer, but checked and also doesnt show up with UI customizer enabled

cp2004, 2022-06-29:
I have no idea why that is not showing on the tab for you then...

cp2004, 2022-12-29:
MPC configuration is now available in 3.3.0rc1, which you can get by switching to 'Release Candidate' in the software update settings.

commit 0a7682ca (2022-06-28): :sparkles: Add M306 MPC temp for hotend

Closes #66