cp2004/OctoPrint-EEPROM-Marlin #67: [Bug]: E0 stepper current not saved

lowerprofile, 2022-07-17:
### Description of the bug

Change the E0 stepper current and hit save. Hit load - the old value is restored. Changing the current at the printer and then loading it shows the coirrect value.

X,Y,Z stepper currents save and load fine.

### Plugin version

3.2.0

### OctoPrint version

1.8.1

### M503 Response

```gcode
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 S0 D1.7500
echo:; Steps per unit:
echo: M92 X80.0000 Y80.0000 Z800.0000 E390.0000
echo:; Maximum feedrates (units/s):
echo:  M203 X500.0000 Y500.0000 Z5.0000 E25.0000
echo:; Maximum Acceleration (units/s2):
echo:  M201 X500.0000 Y500.0000 Z100.0000 E5000.0000
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.0000 R500.0000 T500.0000
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.0000 S0.0000 T0.0000 J0.0800
echo:; Home offset:
echo:  M206 X0.0000 Y0.0000 Z0.0000
echo:; Auto Bed Leveling:
echo:  M420 S1 Z10.0000
echo:  G29 W I0 J0 Z-0.28125
echo:  G29 W I1 J0 Z-0.14625
echo:  G29 W I2 J0 Z0.00625
echo:  G29 W I3 J0 Z0.12875
echo:  G29 W I4 J0 Z0.29250
echo:  G29 W I0 J1 Z-0.23250
echo:  G29 W I1 J1 Z-0.16000
echo:  G29 W I2 J1 Z0.02250
echo:  G29 W I3 J1 Z0.16875
echo:  G29 W I4 J1 Z0.36250
echo:  G29 W I0 J2 Z-0.26250
echo:  G29 W I1 J2 Z-0.13625
echo:  G29 W I2 J2 Z0.05750
echo:  G29 W I3 J2 Z0.20500
echo:  G29 W I4 J2 Z0.42250
echo:  G29 W I0 J3 Z-0.24375
echo:  G29 W I1 J3 Z-0.11625
echo:  G29 W I2 J3 Z0.06375
echo:  G29 W I3 J3 Z0.21875
echo:  G29 W I4 J3 Z0.38000
echo:  G29 W I0 J4 Z-0.19750
echo:  G29 W I1 J4 Z-0.09875
echo:  G29 W I2 J4 Z0.07250
echo:  G29 W I3 J4 Z0.20750
echo:  G29 W I4 J4 Z0.40625
echo:; Material heatup parameters:
echo:  M145 S0 H185.0000 B45.0000 F255
echo:  M145 S1 H240.0000 B110.0000 F255
echo:; PID settings:
echo:  M301 P12.6530 I0.6952 D57.5698
; Controller Fan
echo:  M710 S255 I0 A1 D60 ; (100% 0%)
echo:; Z-Probe Offset (mm):
echo:  M851 X-52.0000 Y0.0000 Z-1.1000
[... 458 lines cut ...]
2022-07-17 07:51:22,552 - octoprint.plugins.eeprom_marlin - INFO - Commands to send: ['M906 E450 X580 Y580 Z580']
2022-07-17 07:51:23,041 - octoprint.plugins.action_command_notification - INFO - Got a notification: Settings Stored
2022-07-17 07:51:23,944 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2022-07-17 07:51:23,977 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 07:51:23,978 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M149 C ; Units in Celsius
2022-07-17 07:51:24,262 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 07:51:24,263 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
2022-07-17 07:51:24,299 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 07:51:24,299 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 X Y Z
2022-07-17 07:51:24,303 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 07:51:24,304 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 T0 E
2022-07-17 07:51:24,324 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI
2022-07-17 07:55:31,309 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 17, 'spoolName': 'Anycubic black PLA', 'material': 'PLA', 'colorName': 'black', 'remainingWeight': 935.4643987152139}' to event-bus
2022-07-17 07:59:46,719 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-07-17 08:00:04,206 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2022-07-17 08:00:04,224 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 08:00:04,225 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M149 C ; Units in Celsius
2022-07-17 08:00:04,523 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 08:00:04,524 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
2022-07-17 08:00:04,560 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 08:00:04,561 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 X Y Z
2022-07-17 08:00:04,565 - octoprint.plugins.eeprom_marlin - WARNING - EEPROM output line not recognized, skipped
2022-07-17 08:00:04,565 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 T0 E
2022-07-17 08:00:04,586 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI
2022-07-17 08:03:06,760 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 17, 'spoolName': 'Anycubic black PLA', 'material': 'PLA', 'colorName': 'black', 'remainingWeight': 935.4643987152139}' to event-bus

cp2004, 2022-07-17:
Currently, the plugin does not support specifying the tool number for those types of commands. It will be trying to send `M569 E<value>`, whereas the firmware will want `M569 T0 E<value>` instead. It requires a more intelligent data structure, I will be working on this and adding support for all the commands that can have different values for different tools, steppers etc. See also #40 and #52.

A couple of days ago I managed to get the native Marlin Simulator up and running, which was an important step in as it means I will be able to test stuff like this where I do not have these features on my printer - it will be coming but I cannot say when I will implement it.

lowerprofile, 2022-07-18:
Thanks for getting back (and thanks for the plugin of course). I have a
tool changer so I use the plugin to save data for the various tool ends, I
just have to remember to set the extruder current at the printer (one setup
has an orbiter with a pancake stepper that cooks at more than 300mA)

I only have the one extruder, is it possible to compile Marlin in such a
way that E0 becomes E? Might have a look later.

cp2004, 2022-12-29:
This is now available through 3.3.0rc1, switch to 'Release Candidate' in the software update settings to get it (might take up to 24  hours to show up)

closed 2023-01-03 by cp2004