cp2004/OctoPrint-EEPROM-Marlin #69: [Request]: Add LONG_FILENAME_WRITE_SUPPORT

jneilliii, 2023-02-04:
### What would you like implemented?

It seems the current Firmware Info doesn't include LONG_FILENAME_WRITE if enabled in Marlin 2.1.2

### M503 Response

```gcode
M503 not relevant, but capabilities report info below...

Cap:LFN_WRITE:1
```
```

cp2004, 2023-02-04:
Hmm, it should just pull the firmware capability report and show everything that is produced there, split on the underscores and capitalised.

[image]

At the bottom - `Lfn Write`. There's no help link as I haven't specifically added one, but it should show there.

jneilliii, 2023-02-04:
Apologies, I do see that now,

closed 2023-02-04 by jneilliii